OctoPrint/OctoPrint #4963: downloadable font: Glyph bbox was incorrect

gatonero, 2024-03-02:
### The problem

Starting Octoprint  in safe mode with dvelopment console shows warning related to downloadable font Glyph bbox.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

octopi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240302033635.zip](https://github.com/OctoPrint/OctoPrint/files/14467795/octoprint-systeminfo-20240302033635.zip)

downloadable font: Glyph bbox was incorrect (glyph ids 1 3 4 5 6 7 9 12 22 24 25 29 30 33 34 35 36 37 39 40 44 45 51 52 54 56 58 59 60 73 85 94 95 109 112 123 132 133 137 147 155 165 175 178 187 197) (font-family: "Font Awesome 6 Free" style:normal weight:400 stretch:100 src index:0) source: https://octopi.local/static/vendor/fontawesome-6.1.1/webfonts/fa-regular-400.woff2

downloadable font: Glyph bbox was incorrect (glyph ids 47 69 76 77 95 96 97 98 101 102 103 104 126 133 134 137 153 173 176 178 180 192 198 199 231 232 286 287 288 289 293 295 298 304 305 306 323 324 333 335 337 338 340 343 344 345 346 347 348 353 361 365 366 367 368 371 372 380 381 384 385) (font-family: "FontAwesome3" style:normal weight:400 stretch:100 src index:1) source: https://octopi.local/static/vendor/font-awesome-3.2.1/fonts/fontawesome-webfont.woff?v=3.2.1

downloadable font: Glyph bbox was incorrect (glyph ids 1 3 4 5 6 7 9 12 22 24 25 29 30 33 34 35 36 37 38 39 45 46 48 49 50 51 53 54 55 58 60 64 65 66 67 71 72 73 81 86 87 88 91 93 100 101 105 106 109 112 113 130 133 140 141 142 146 148 149 151 153 154 155 162 165 170 175 180 181 188 191 193 194 195 196 198 199 201 204 205 212 213 215 216 221 223 225 227 228 232 243 245 246 248 250 251 254 255 257 258 261 262 266 269 270 273 275 276 277 280 281 284 287 291 292 295 300 303 306 307 308 313 314 316 319 321 324 325 327 328 330 334 335 336 337 342 345 348 349 351 357 359 360 362 366 369 371 372 373 374 376 377 381 384 387 388 389 396 398 399 400 401 416 417 420 425 426 428 432 433 437 439 440 441 444 449 454 456 457 458 461 464 469 470 471 472 475 477 480 481 483 485 486 487 488 490 491 496) (font-family: "Font Awesome 6 Brands" style:normal weight:400 stretch:100 src index:0) source: https://octopi.local/static/vendor/fontawesome-6.1.1/webfonts/fa-brands-400.woff2

Die Ressource unter "https://octopi.local/static/vendor/fontawesome-6.1.1/webfonts/fa-v4compatibility.woff2" wurde vorab geladen ("link-preload"), aber nicht innerhalb der ersten Sekunden verwendet. Es wird eine Überprüfung empfohlen, ob alle preload-Attribute korrekt gesetzt sind.

foosel, 2024-03-11:
This is an issue with Font Awesome and current versions of Firefox, see FortAwesome/Font-Awesome#19925. Apparently it has been solved in Font Awesome 6.5, so bundling that with OctoPrint should fix this.

foosel, 2024-03-12:
Fixed in 4b2c6675d5886f174a7b0657723884000580ea79 (forgot to tag this issue). Will be in 1.11.0 (this came too late for 1.10.0).

However, there will still be a glyph warning about `font-awesome-3.2.1` since that isn't fixed by Font Awesome and we still need to include this regardless to stay backwards compatible to third party plugins that haven't yet updated.

foosel, 2025-04-22:
Closing this as 1.11.0 has just been released

closed 2025-04-22 by foosel