OctoPrint/OctoPrint #4965: Out of the blue, connecting to /dev/ttyACM0 fails, but AUTO works (to /dev/ttyACM0)

thomergil, 2024-03-11:
### The problem

Out of nowhere, OctoPrint no longer connects to my Prusa MK3S+, FW 3.13.2 when the serial port is set to `/dev/ttyACM0`. When I set it to `AUTO` it works, but it connects to `/dev/ttyACM0`. Baud rate is confirmed correct. The same problem happens in Safe Mode. I checked and unchecked "Request exclusive access to the serial port" -- no change.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

Version 1.9.3

### Operating system running OctoPrint

Linux

### Printer model & used firmware incl. version

Prusa MK3S+, FW 3.13.2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

On failure:
```
==> octoprint.log <==
2024-03-10 23:49:14,137 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2024-03-10 23:49:14,345 - octoprint.server - INFO - Autorefresh of serial port list stopped
```
and `serial.log`:
```
==> serial.log <==
2024-03-10 23:59:57,334 - Recv: T:25.2 /0.0 B:23.9 /0.0 T0:25.2 /0.0 @:0 B@:0 P:0.0 A:29.9
2024-03-10 23:59:59,341 - Recv: T:25.4 /0.0 B:24.1 /0.0 T0:25.4 /0.0 @:0 B@:0 P:0.0 A:30.3
2024-03-11 00:00:01,344 - Recv: T:25.2 /0.0 B:24.3 /0.0 T0:25.2 /0.0 @:0 B@:0 P:0.0 A:29.7
2024-03-11 00:00:03,351 - Recv: T:25.3 /0.0 B:24.0 /0.0 T0:25.3 /0.0 @:0 B@:0 P:0.0 A:29.7
2024-03-11 00:00:05,354 - Recv: T:25.2 /0.0 B:23.9 /0.0 T0:25.2 /0.0 @:0 B@:0 P:0.0 A:29.8
```
and then it hangs until I press the "Disconnect" button.

When connecting to AUTO:
```
==> serial.log <==
2024-03-11 00:00:15,539 - Changing monitoring state from "Offline" to "Detecting serial connection"

==> octoprint.log <==
2024-03-11 00:00:15,539 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"

==> serial.log <==
2024-03-11 00:00:15,572 - Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200

==> octoprint.log <==
2024-03-11 00:00:15,572 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200

==> serial.log <==
2024-03-11 00:00:15,572 - Trying port /dev/ttyACM0, baudrate 115200
2024-03-11 00:00:15,574 - Connecting to port /dev/ttyACM0, baudrate 115200

==> octoprint.log <==
2024-03-11 00:00:15,573 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyACM0, baudrate 115200

==> tornado.log <==
2024-03-11 00:00:15 - tornado.access - INFO - 204 POST /api/connection (::ffff:192.168.1.5) 257.89ms
2024-03-11 00:00:15 - tornado.access - INFO - 200 POST /api/connection (::ffff:192.168.1.5) 258.53ms

==> octoprint.log <==
2024-03-11 00:00:15,573 - octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 115200
2024-03-11 00:00:15,616 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s

==> serial.log <==
2024-03-11 00:00:15,615 - Handshake attempt #1 with timeout 2.0s
2024-03-11 00:00:15,624 - Connected to: Serial<id=0xdf45d250, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2024-03-11 00:00:15,651 - Send: N0 M110 N0*125
2024-03-11 00:00:15,657 - Recv: ok

==> octoprint.log <==
2024-03-11 00:00:15,650 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0

==> serial.log <==
2024-03-11 00:00:15,680 - Changing monitoring state from "Detecting serial connection" to "Operational"

==> octoprint.log <==
2024-03-11 00:00:15,680 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Operational"
2024-03-11 00:00:15,709 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0

==> serial.log <==
2024-03-11 00:00:15,710 - Send: N0 M110 N0*125

==> octoprint.log <==
2024-03-11 00:00:15,776 - octoprint.server - INFO - Autorefresh of serial port list stopped
2024-03-11 00:00:15,778 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2024-03-11 00:00:15,781 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
```

[octoprint-systeminfo-20240310235800.zip](https://github.com/OctoPrint/OctoPrint/files/14553340/octoprint-systeminfo-20240310235800.zip)

thomergil, 2024-03-11:
This may be a red herring. It may be a printer issue, not an Octoprint issue. I have the same problem with PrusaSlicer, when trying to upgrade the firmware.

thomergil, 2024-03-11:
I confirmed this problem went away after replacing the Einsy board. It was not an OctoPrint problem.

closed 2024-03-11 by thomergil

foosel, 2024-03-11:
Glad to hear you got this sorted out, thanks for reporting back :)