OctoPrint/OctoPrint #4971: [RC Feedback] Feedback on the 1.10.0rc3 Release Candidate

foosel, 2024-03-18:
Please provide *general* feedback on your experience with the 1.10.0rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- User and group management functioning as expected.
- Plugin installation functioning as expected.
- Application key management functioning as expected. Authentication workflow with third party clients at your disposal (e.g. slicers) works as it should.
- Backup creation, download and restore functioning as expected

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
###
### ❗👆 IMPORTANT 👆❗

---

### Update test matrix

[image]

*Releases aren't rolled out unless all tests in the update test matrix succeed.*

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

- zeroconf dependency gets built from source instead of pulled from piwheels on OctoPi 0.18, see https://github.com/OctoPrint/OctoPrint/issues/4938#issuecomment-1919869375
  - caused by an outdated pip version, [see this comment](https://github.com/OctoPrint/OctoPrint/issues/4938#issuecomment-1921262160)
  - if affected (OctoPi 0.18.0 development build instead of stable?), make sure to update to `pip>=20.3`
- UI Customizer Plugin: LazeMSS/OctoPrint-UICustomizer#315 - solved with release 0.1.9.91
  - 12-column layout no longer fits
  - camera has additional letterbox

### Issues needing analysis

to-str, 2024-03-18:
ran rc2 extensively on octopi (debian 11.9) without problems (only thing that bugged me are the achievements, this is a tool, not a toy, but 'bugging me' and 'bug' are not the same thing ;)), updated to rc3 and started a 5h print, so far so good.

foosel, 2024-03-19:
@to-str 

The achievements are a way to remind people that running this project takes time and effort and requires funding, at a time when they actually measurably gain value from it. It is a way to ensure the future funding of this project, while still also giving people back something instead of just plastering them with banners left and right.

My cry for help in October resulted in a ton of feedback along the lines of "omg, I had no idea" - this is my attempt in giving people a hint at how this project stays afloat without being too invasive and annoying, in the shape of functionality they might actually enjoy.

Whoever doesn't like that can just disable the plugin, that's the beauty of it.

to-str, 2024-03-19:
@foosel @jneilliii Oh thanks, I could swear last time I checked the disable was greyed out. But maybe I had a print running. By now I probably got through most of the achievements that I am going to get.

The achievements are fine, by all means, just not my thing, I am easily distracted (and I thought they could not be disabled). Your project, your rules, I am essentially just a 'nutzniesser' here :) Other than that, I am two long-ish prints in (5h each) and no problems so far whatsoever.

fredrikbaberg, 2024-03-25:
Temperature in UI (bed and tool) doesn't automatically update for me, I have to refresh the browser window. It's either kept at the previous value or just left at zero.

This occurs on the printer tab in OrcaSlicer as well as if I open OctoPrint in Microsoft Edge Version 122.0.2365.92 (Officiell version) (64 bitar).

I have not yet tested it in safe mode, as I'm currently printing and just now noticed it.

trunzoc, 2024-03-31:
FWIW, I love the quest for the hidden achievements.

That being said, I think Heavy Chonker is broken.  I've repeatedly uploaded 500mb+ gcode files and haven't received the achievement yet.

((11 more hidden achievements to find)

foosel, 2024-04-02:
> I think Heavy Chonker is broken

Ah, yeah, confirmed... I'm testing for a size info from the event payload there that apparently I forgot to add. Consider this achievement *very* hidden then, will be fixed in an upcoming bugfix release (while yes, it is a bug, it's not something I'd say is bad enough that we need to do another RC round given how much overhead that causes).

JohnOCFII, 2024-04-02:
OctoPrint 1.10.0rc3 on OctoPi 1.0.0 seems to be working well for me:

Upgraded from 1.10.0rc1.

Uploaded new GCODE to OctoPrint
Printed multiple GCODE files on Prusa MK4
GCODE viewer worked well
Terminal worked well
Timelapse worked well
OctoSlack plugin worked well
Print Job History plugin worked well
Announcements/Notifications in title bar worked well
Classic Webcam and Camera settings all worked well
Achievements Plug-in is fun!

foosel, 2024-04-02:
I fear we might need another RC round... while helping test a recent update to the FileCheck plugin, @b-morgan ran into some issues switching the plugin to its prerelease channel (OctoPrint/OctoPrint-FileCheck#13) that on closer inspection look very much like a bug in the performance improvements on the settings in 1.10.0. I'm still investigating, but if this is indeed the case, this means we'll need 1.10.0rc4 after all (which I as very confident we would be able to go without until around two hours ago).

*sigh*

ChrisHeerschap, 2024-04-04:
Been running 1.10 rc3 on two of my three instances. One is still on 1.9.3 because I never got around to switching it to the RC channel, but the good news is I had to check which two were running the RC because I couldn't tell. Been using all three pretty regularly since the 1.10 and can't say I've noticed any problems. Buncha little annoyances that I don't think are related and aren't annoying enough to really dig into what's causing them. 

The most concerning problems I've had are I believe hardware related and are on the 1.9.3 system anyway.

foosel, 2024-04-11:
1.10.0rc4 was released a few days ago, new feedback ticket is #4988

closed 2024-04-11 by foosel