OctoPrint/OctoPrint #4973: Debug run eats errors

smurfix, 2024-03-21:
### The problem

1. Subtly break your installation while working on an extension
2. Run `python3 -moctoprint serve --debug`
3. Hit a fatal error on startup
4. Observe no stack trace, despite `--debug`

IMHO, requiring people to hack source code just to get useful error information is not acceptable behavior.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

current master

### Operating system running OctoPrint

not relevant

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

```
$ python3 -moctoprint serve --debug --port 50081
[ lines about timing elided ]
2024-03-21 08:25:33,078 - octoprint.startup - CRITICAL - Could not initialize logging: Unable to configure formatter 'colored'
2024-03-21 08:25:33,079 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
Could not initialize logging: Unable to configure formatter 'colored'
There was a fatal error starting up OctoPrint.
```

smurfix, 2024-03-21:
No, a bundle is not required to process this. _Sigh_

foosel, 2024-03-25:
Agreed, that needs looking into.

It would help if you could explain what exactly you did that broke it that way so I can actually reproduce this error case and then fix the behaviour here.

---

Also, could you please turn down the grumpiness a bit? Bugs happen, it's not like that this behaviour is intentional, so assuming that "requiring people to hack source code just to get useful error information" is the position here is just unfair.

And most people reporting bugs here are *not* capable to decide whether a bundle is needed or not, so yes, the bot is annoying, but the alternative is having to spend even more time running after information needed to actually fix bugs, rather than spending it on fixing them. If you have a better idea how to make this work without a bot that can cause the one or other false positive, I'm all ears, because I'm not exactly happy with it either but it's the only thing I found that at least somewhat works.

foosel, 2024-03-28:
Fixed by the above commits, to be included in 1.11.0.

Example:

```
2024-03-28 19:34:47,318 - octoprint.startup - CRITICAL - There was a fatal error initializing OctoPrint:
Traceback (most recent call last):
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/__init__.py", line 104, in init_platform
    logger = init_logging(
             ^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/__init__.py", line 265, in init_logging
    raise RuntimeError("Everything is broken")
RuntimeError: Everything is broken

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/cli/server.py", line 112, in run_server
    components = init_platform(
                 ^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/__init__.py", line 116, in init_platform
    raise FatalStartupError("Could not initialize logging", cause=ex)
```

Also improved the logging in case of invalid YAML:

```
$ octoprint config get accessControl.autologinAs
Traceback (most recent call last):
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/settings/__init__.py", line 912, in load
    config = yaml.load_from_file(file=f)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/util/yaml.py", line 28, in load_from_file
    return yaml.load(file, Loader=SafeLoader)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "yaml/_yaml.pyx", line 673, in yaml._yaml.CParser.get_single_node
  File "yaml/_yaml.pyx", line 687, in yaml._yaml.CParser._compose_document
  File "yaml/_yaml.pyx", line 731, in yaml._yaml.CParser._compose_node
  File "yaml/_yaml.pyx", line 847, in yaml._yaml.CParser._compose_mapping_node
  File "yaml/_yaml.pyx", line 860, in yaml._yaml.CParser._parse_next_event
yaml.scanner.ScannerError: mapping values are not allowed in this context
  in "/home/gina/.octoprint/config.yaml", line 169, column 13

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/__init__.py", line 206, in init_settings
    return settings(
           ^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/settings/__init__.py", line 83, in settings
    _instance = Settings(
                ^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/settings/__init__.py", line 549, in __init__
    self.load(migrate=True)
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/settings/__init__.py", line 916, in load
    raise InvalidYaml(self._configfile, error=str(e))
octoprint.settings.InvalidYaml: Error parsing the configuration file /home/gina/.octoprint/config.yaml, it is invalid YAML: mapping values are not allowed in this context
  in "/home/gina/.octoprint/config.yaml", line 169, column 13

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/cli/config.py", line 73, in cli
    ctx.obj.settings = init_settings(
                       ^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/__init__.py", line 210, in init_settings
    raise FatalStartupError(str(e))
octoprint.FatalStartupError: Error parsing the configuration file /home/gina/.octoprint/config.yaml, it is invalid YAML: mapping values are not allowed in this context
  in "/home/gina/.octoprint/config.yaml", line 169, column 13

There was a fatal error initializing the settings manager: Error parsing the configuration file /home/gina/.octoprint/config.yaml, it is invalid YAML: mapping values are not allowed in this context
  in "/home/gina/.octoprint/config.yaml", line 169, column 13
```

commit 5bde1dff (2024-03-28): 🐛 Include stack trace with fatal errors

Also make InvalidYaml exceptions more helpful by including
the full message from the triggering YAMLError.

Closes #4973