OctoPrint/OctoPrint #4976: Files List sometimes forgets saved sort order

timur-tabi, 2024-03-26:
### The problem

The Files list window (on the left side) has an option to specify the sort order.  The top most entry is "Sort by name (ascending)".  

I have set this option to "Sort by upload date (descending)".  99% of the time, when I start Octoprint, this setting is retained.  But every now and then, the menu resets to "Sort by name (ascending)".  

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

OctoPi 1.0.0cam

### Browser and version of browser, operating system running browser

Happens in both Firefox and Chrome for Linux

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240325221755.zip](https://github.com/OctoPrint/OctoPrint/files/14751963/octoprint-systeminfo-20240325221755.zip)

cp2004, 2024-04-06:
These settings are saved to your browser's local storage, they aren't global configuration. Do you have any extensions/settings that might block or clean up cookies/saved data from websites periodically? Sometimes, the local storage data is rolled in with cookie/privacy settings, even though it is not actually cookies at all.

timur-tabi, 2024-04-07:
Not that I know of, but I will try to remember to pay attention next time I do something along the lines of deleting cookies.

I don't understand why this configuration is stored in the browser.  Why not store in in config.yaml like everything else?

foosel, 2024-04-07:
Because you might want to have different sorting options per browser. E.g. desktop vs a browser on a tablet mounted right above your printer. And it's a UI thing anyhow (the code running in the browser is doing the sorting, not the server).

timur-tabi, 2024-04-07:
Hmmm.. fair enough.

closed 2024-04-07 by timur-tabi

cp2004, 2024-04-07:
Why did you close this issue? Did you work out that it was the browser clearing the data?

timur-tabi, 2024-04-07:
I closed it because it'll probably be a long time before I can reproduce this issue, and I didn't think it would be fair to keep it open until then.