OctoPrint/OctoPrint #4977: [Request] Prusa XBuddy: handling auto-pause

smurfix, 2024-03-28:
### Is your feature request related to a problem? Please describe.

When the printer goes into Paused state due to some problem (nozzle dirty, failed autolevel), it disables its heaters until explicitly told to continue via M602 or M603.

Restarting the print job thus doesn't work because the printer goes through all the motions but of course nothing gets extruded. Also, setting the temperature in Octoprint does nothing.

### Describe the solution you'd like

What Octoprint should do to mitigate this situation is to monitor the temperature (both current and target) after a "successful" M140/M190. If the real temperature is way off, or the target temperature as reported by M155 doesn't change, the print should be stopped with an error.

### Describe alternatives you've considered

A possible workaround is to add a "blind" M603 to OctoPrint's start code, but that's on the user. It's also not obvious that this is necessary because the Prusa unhelpfully doesn't *say* that it's paused on its display.

### Additional context

Maybe the "heating doesn't work" error message could point the user to the possible necessity to send a M603 (added to the start gcode, manually, …).

jneilliii, 2024-03-28:
This seems more like a firmware bug that should be reported to Prusa.

smurfix, 2024-03-28:
I did that too.

However, ultimately the problem is that the printer reported to Octoprint that it's paused, i.e. it behaves as if it received a M601. If Octoprint wants to continue or restart printing it might be expected to un-pause the thing.

Simply starting to print is obviously not sufficient, and a quick glance over the g-code list on the RepRap wiki doesn't yield clues as to which other commands should, or should not, auto-un-pause.

foosel, 2024-03-28:
Can you please share a serial.log of such a situation?

smurfix, 2024-03-29:
> Can you please share a serial.log of such a situation?

Sure.

[serial.log.zip](https://github.com/OctoPrint/OctoPrint/files/14801464/serial.log.zip)

foosel, 2024-04-11:
For the record, that firmware doesn't just not say that it's still in a paused state on the display as you say, there's also no way to see that on the serial communication. And frankly, the reset of the target happens SO late after the initial pause that it rather looks like a bug in whatever else it is going through here at the start rather than anything intentional. Annotated `serial.log` follows

```
// something fails during probing, firmware sends `action:pause` to the printer

2024-03-28 20:38:57,622 - Recv: echo:Probe classified as NOK (low-precision)
2024-03-28 20:38:57,623 - Recv: echo:enqueueing "M118 A1 action:pause"
2024-03-28 20:38:57,624 - Recv: ok
2024-03-28 20:38:57,625 - Recv: // action:pause
2024-03-28 20:38:57,626 - Pausing on request of the printer...
2024-03-28 20:38:57,627 - Changing monitoring state from "Printing" to "Pausing"
2024-03-28 20:38:57,629 - Send: N38 G0 Z5 F480*7
2024-03-28 20:38:57,653 - Recv: ok
2024-03-28 20:38:57,678 - Send: N39 M400*29
2024-03-28 20:38:57,969 - Recv:  T:184.00/185.00 B:85.10/85.00 C:-30.00/0.00 X1:31.00/36.00 A:67.68/0.00 T0:20.00/0.00 T1:184.00/185.00 T2:21.00/0.00 T3:21.00/0.00 T4:20.00/0.00 T5:6.00/0.00 @:37 B@:0 HBR@:51 @0:0 @1:37 @2:0 @3:0 @4:0 @5:0 B_0_0:20.80/0.00 B_1_0:20.30/0.00 B_2_0:20.10/0.00 B_3_0:20.10/0.00 B_0_1:20.60/0.00 B_1_1:23.70/0.00 B_2_1:27.00/0.00 B_3_1:26.80/0.00 B_0_2:21.90/0.00 B_1_2:42.50/42.50 B_2_2:63.80/63.70 B_3_2:63.80/63.70 B_0_3:23.10/0.00 B_1_3:63.80/63.70 B_2_3:85.10/85.00 B_3_3:85.20/85.00

// At this point `T1` is still at a target of 185degC

// [...]

2024-03-28 20:39:07,312 - Send: N40 M114*19
2024-03-28 20:39:07,320 - Recv: X:42.13 Y:-7.41 Z:5.00 E:-3.00 Count A:2800 B:3919 Z:3885
2024-03-28 20:39:07,354 - Recv: ok
2024-03-28 20:39:07,363 - Changing monitoring state from "Pausing" to "Paused"

// now OctoPrint has processed the pause

2024-03-28 20:39:07,967 - Recv:  T:185.00/185.00 B:85.10/85.00 C:-30.00/0.00 X1:31.00/36.00 A:67.85/0.00 T0:20.00/0.00 T1:185.00/185.00 T2:21.00/0.00 T3:21.00/0.00 T4:20.00/0.00 

// T1 is still sitting at a stable 185degC

// [...]

2024-03-28 20:39:33,965 - Recv:  T:185.00/185.00 B:85.20/85.00 C:-30.00/0.00 X1:30.00/36.00 A:68.67/0.00 T0:20.00/0.00 T1:185.00/185.00 T2:22.00/0.00 T3:21.00/0.00 T4:19.00/0.00 T5:6.00/0.00 @:38 B@:0 HBR@:51 @0:0 @1:38 @2:0 @3:0 @4:0 @5:0 B_0_0:21.30/0.00 B_1_0:20.30/0.00 B_2_0:20.10/0.00 B_3_0:20.20/0.00 B_0_1:20.70/0.00 B_1_1:23.90/0.00 B_2_1:27.50/0.00 B_3_1:27.30/0.00 B_0_2:22.10/0.00 B_1_2:42.50/42.50 B_2_2:63.70/63.70 B_3_2:63.70/63.70 B_0_3:23.50/0.00 B_1_3:63.70/63.70 B_2_3:85.20/85.00 B_3_3:85.10/85.00

// Print gets resumed

2024-03-28 20:39:34,671 - Changing monitoring state from "Paused" to "Starting"
2024-03-28 20:39:34,736 - Send: N0 M110 N0*125
2024-03-28 20:39:34,743 - Recv: ok
2024-03-28 20:39:34,747 - Changing monitoring state from "Starting" to "Printing"
2024-03-28 20:39:34,806 - Send: N1 M73 P0 R17*34
2024-03-28 20:39:34,814 - Recv: ok
2024-03-28 20:39:34,817 - Send: N2 M201 X7000 Y7000 Z200 E2500*9
2024-03-28 20:39:34,831 - Recv: ok
2024-03-28 20:39:34,854 - Send: N3 M203 X450 Y450 Z12 E100*13
2024-03-28 20:39:34,860 - Recv: ok
2024-03-28 20:39:34,872 - Send: N4 M204 P4000 R1200 T5000*85
2024-03-28 20:39:34,877 - Recv: ok
2024-03-28 20:39:34,881 - Send: N5 M205 X8.00 Y8.00 Z2.00 E10.00*12
2024-03-28 20:39:34,905 - Recv: ok
2024-03-28 20:39:34,963 - Send: N6 M205 S0 T0*37
2024-03-28 20:39:35,052 - Recv: ok
2024-03-28 20:39:35,089 - Send: N7 M17*18
2024-03-28 20:39:35,163 - Recv: ok
2024-03-28 20:39:35,165 - Send: N8 M862.3 P "XL"*126
2024-03-28 20:39:35,574 - Recv: ok
2024-03-28 20:39:35,585 - Send: N9 M862.5 P2*127
2024-03-28 20:39:35,605 - Recv: ok
2024-03-28 20:39:35,609 - Send: N10 M862.6 P"Input shaper"*29
2024-03-28 20:39:35,662 - Recv: ok
2024-03-28 20:39:35,683 - Send: N11 M115 U5.1.2+13478*71
2024-03-28 20:39:35,734 - Recv: FIRMWARE_NAME:Prusa-Firmware-Buddy 5.1.2+13478 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa-XL EXTRUDER_COUNT:6 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2024-03-28 20:39:35,735 - Recv: Cap:SERIAL_XON_XOFF:0
2024-03-28 20:39:35,736 - Recv: Cap:BINARY_FILE_TRANSFER:0
2024-03-28 20:39:35,736 - Recv: Cap:EEPROM:0
2024-03-28 20:39:35,738 - Recv: Cap:VOLUMETRIC:1
2024-03-28 20:39:35,739 - Recv: Cap:AUTOREPORT_TEMP:1
2024-03-28 20:39:35,740 - Recv: Cap:PROGRESS:0
2024-03-28 20:39:35,741 - Recv: Cap:PRINT_JOB:1
[... 283 lines cut ...]
// T1 gets selected and target temperature gets set to 0 again
// we are now at T+2min since the print got paused initially and it sat at 
// target most of that time and only now things broke

2024-03-28 20:40:47,371 - Send: N52 T1 S1 L0 D0*70
2024-03-28 20:40:48,094 - Recv:  T:144.00/0.00 B:85.20/85.00 C:-30.00/0.00 X1:30.00/36.00 A:71.14/0.00 T0:20.00/0.00 T1:144.00/0.00 T2:21.00/0.00 T3:21.00/0.00 T4:21.00/0.00 T5:6.00/0.00 @:0 B@:0 HBR@:51 @0:0 @1:0 @2:0 @3:0 @4:0 @5:0 B_0_0:21.80/0.00 B_1_0:20.90/0.00 B_2_0:20.40/0.00 B_3_0:20.40/0.00 B_0_1:21.00/0.00 B_1_1:24.40/0.00 B_2_1:28.40/0.00 B_3_1:28.10/0.00 B_0_2:22.70/0.00 B_1_2:42.50/42.50 B_2_2:63.80/63.70 B_3_2:63.70/63.70 B_0_3:24.50/0.00 B_1_3:63.70/63.70 B_2_3:85.20/85.00 B_3_3:85.10/85.00
2024-03-28 20:40:48,857 - Recv: ok
2024-03-28 20:40:48,860 - Send: N53 G92 E0*113
2024-03-28 20:40:48,870 - Recv: X:94.23 Y:359.59 Z:10.95 E:0.00 Count A:36328 B:-21272 Z:9690
2024-03-28 20:40:48,872 - Recv: ok

[... cut ...]  decides to go out of their way to make things incompatible with existing software, that's their issue, not mine. Sorry. I know as a user you are between a rock and hard place here with regards to that, but I've really had it with seeing the simplest things getting broken by that company in particular these days, established standards, commands and best practices getting ignored (feels like "not invented here" syndrome) and I then getting bombarded with requests to fix that somehow in OctoPrint.

closed 2024-04-11 by foosel

smurfix, 2024-04-11:
Fair enough, and thanks a lot for going above and beyond your job, annotating this serial log! ❣️