OctoPrint/OctoPrint #4978: GCode Viewer is disabled after re-upload

smurfix, 2024-03-28:
### The problem

* upload a file to OctoPrint via Prusa Slicer
* load file
* observe nice gcode preview
* re-upload the same file, or upload a different file with the same name
* OctoPrint updates the data (size etc.) of the loaded file
* … but the gcode viewer is now empty.

To recover, I need to reload the browser window.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10-0rc3

### Operating system running OctoPrint

Debian

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240328221240.zip](https://github.com/OctoPrint/OctoPrint/files/14795707/octoprint-systeminfo-20240328221240.zip)

foosel, 2024-04-02:
First thought this was a regression, but I can also reproduce this on 1.9.3, so no regression on 1.10.0 and thus will have to go into a bugfix release.

foosel, 2024-04-03:
Good news I guess, we need another RC for 1.10.0 after all, and I'll include this fix in that.

foosel, 2024-04-11:
Fixed in 1.10.0rc4.

commit 6a9239fc (2024-04-02): 🐛 GCV: Fix reload of same file

We need to reload if either the name or the size or the date differ,
not only if they ALL differ.

Closes #4978
closed 2024-04-11 by foosel