OctoPrint/OctoPrint #4981: Streamer_select.service stopped causing lagging webcam

WhiteMaverick, 2024-03-30:
### The problem

The streamer_select.service is stopped on nightly build 2024-03-28_2024-03-15-octopi-bookworm-arm64-lite-1.1.0.zip
The octopi.txt file is missing from /boot causing the service to fail.
copying /boot/firmware/octopi.txt to /boot fixes the issue.

[octoprint-systeminfo-20240329215656.zip](https://github.com/OctoPrint/OctoPrint/files/14809890/octoprint-systeminfo-20240329215656.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

1.1.0

### Printer model & used firmware incl. version

monoprice i3 mini

### Browser and version of browser, operating system running browser

firefox 124.0.1 64 bit

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

I also configured the webcam in octopi.txt as follows:
camera="usb"
camera_usb_options="-r 1920x1080 -f 30 -d /dev/v4l/by-id/usb-046d_HD_Pro_Webcam_C920_5DA9ABAF-video-index0"

Then the webcam worked as intended.
The webcam issue has existed in the past few nightly builds that I have tried on my pi5 8GB

jneilliii, 2024-03-30:
This should be reported to the OctoPi project, not OctoPrint.

https://github.com/guysoft/OctoPi

WhiteMaverick, 2024-03-30:
sorry, my mistake

closed 2024-03-30 by WhiteMaverick