OctoPrint/OctoPrint #4984: "Mass production" achievement ignores uploads

smurfix, 2024-04-05:
### The problem

When you do a test print / tweak dimensions / repeat cycle and re-use the file name, that shouldn't count as mass production.

Thus
1. upload file named X
2. print X
3. upload different file named X
4. print the new X
5. repeat eight more times

shouldn't yield an achievement. Or, well, maybe it should give you a different achievement, like "ecologist" because you're recycling (file names, that is …), but not the "mass production" one.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0rc3

### Operating system running OctoPrint

Debian

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

yes I know, the version number in the bundle is still wrong.

[octoprint-systeminfo-20240328221240.zip](https://github.com/OctoPrint/OctoPrint/files/14884041/octoprint-systeminfo-20240328221240.zip)

jneilliii, 2024-04-05:
Probably easy to adjust to use file hash instead of name here for calculation?

foosel, 2024-04-08:
I've now added the timestamp of the last modification to the check. Ready for 1.10.0rc4

foosel, 2024-04-11:
Fixed in 1.10.0rc4.

commit 6221dec6 (2024-04-09): 🐛 Add time to check to Mass Production achievement

Modifications of the file should reset the counter.

Closes #4984
closed 2024-04-11 by foosel