OctoPrint/OctoPrint #4985: Does not reconnect properly when serial port reappears, even after refresh

timur-tabi, 2024-04-05:
### The problem

I upgraded my Prusa MK3s to an MK3.5, which includes a new control board that has a USB-C instead of a USB-B port.  Octoprint works fine once connected.

With the new MK3.5 board, when the printer is off, the /dev/ttyACM0 serial port disappears.  Octoprint detects this and posts a warning:

`No serial port found, are you sure your printer is physically connected and supported? Try [refreshing](javascript:void(0)) and if that doesn't help please see [the FAQ](https://faq.octoprint.org/no-candidates).`

When the printer is turned on, Octoprint detects that the serial port has returned.  The Terminal console shows:

`Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyACM0, baudrate 230400`

However, Octoprint does not actually connect.  The "Connect button" on the left side changes to "Disconnect", so Octoprint thinks it's connected, but there is no actual serial communication happening.  The buttons on the Control page are disabled, so the rest of Octoprint is aware that it is not connected.

In order for me to actually connect to the serial port, I need to click on "Disconnect" and then click on "Connect".

Turning the printer off shows this on the Terminal:

`Changing monitoring state from "Opening serial connection" to "Offline"`

I would attach a "serial.log", but when I enable serial port logging, the behavior changes (I get a new error pop-up and Octoprint gets more confused about the error).  Also, the serial log only contains the same thing I wrote above.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

In safe mode, nothing appears in the Terminal.  It does appear that auto-reconnect is a feature that is not present in safe mode.

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

OctoPi 1.0.0cam

### Printer model & used firmware incl. version

Prusa MK3.5

### Browser and version of browser, operating system running browser

Firefox for Linux

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240405095941.zip](https://github.com/OctoPrint/OctoPrint/files/14886648/octoprint-systeminfo-20240405095941.zip)

timur-tabi, 2024-04-06:
I installed the PortLister plugin, and surprisingly, it has no effect.  Octoprint still performs only a partial connect when the printer is turned on.

timur-tabi, 2024-04-06:
Here''s a freshly booted system with the printer off:

[image]

and here it is after I turn the printer on. 

[image]

You can see that the Connect button now says "Disconnect".  When I ssh into the Pi, I can see that /dev/ttyACM0 does exist now.  I have to click on Disconnect and then Connect to actually connect to the printer.

cp2004, 2024-04-06:
OctoPrint does not try and automatically connect to the printer itself. You've seen this as you said this behaviour did not exist in safe mode. This likely means that there is a plugin that is starting this connection attempt.

It is easiest if you run all the troubleshooting in safe mode to rule out 3rd party plugins. Do you see any issues connecting to your printer there? When you click connect after the printer first powers on, does it work fine?

It could be that the plugin that is trying to auto connect is doing so too quickly for the system/printer to respond, and it needs a longer delay configured.

timur-tabi, 2024-04-07:
Ah, indeed.  It's the TPLINK plugin that takes care of this, and I had to change the connect delay from 2 to 20 seconds.

Sorry for the noise.  It's hard to tell sometimes what features are Octoprint and what is from a plugin, especially if you have a lot of plugins and you haven't touched them in months.

closed 2024-04-07 by timur-tabi

foosel, 2024-04-08:
> It's hard to tell sometimes what features are Octoprint and what is from a plugin, especially if you have a lot of plugins and you haven't touched them in months.

Which is why we always want people to test in safe mode, as that disables all third party plugins 😉