OctoPrint/OctoPrint #4986: [Request] Virtual Printer Warning

foxt, 2024-04-07:
### Is your feature request related to a problem? Please describe.

There has been many times where I have been confused why actions I make inside OctoPrint aren't reflected on the physical printer, the reson for this is the 'virtual' printer is in use, as occasionally OctoPrint defaults to the virtual printer when connecting instead of the actual printer. This would be even more problematic if I didn't have a camera on my OctoPrint instance and could not see that the printer was doing nothing.

### Describe the solution you'd like

A warning that appears when connected to the Virtual printer.

[image]

### Describe alternatives you've considered

Alternatively, make sure that OctoPrint will never default to the VIRTUAL printer when using  'AUTO', or even separating the options
[image]

jneilliii, 2024-04-07:
Do you use the virtual printer? If not just disable it and that won't happen.

foosel, 2024-11-18:
Closing due to lack of response