OctoPrint/OctoPrint #4987: [1.10.0rc3] No such file or directory: 'build\\lib\\src/octoprint\\_static_version.py' when installing on Windows

MaleNurse, 2024-04-08:
### The problem

When installing clean or updating OctoPrint on a Windows machine, the install errors out on "No such file or directory: 'build\\lib\\src/octoprint\\_static_version.py'"

I've tried this both by trying to update a v1.9.3 install on Windows Server 2022 running Python 3.11 as well as installing clean on a Win10 box running Python 3.12. This also started w/ the first rc for 1.10.0.

I've included the output of the install in the files section below.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0rc3

### Operating system running OctoPrint

Windows 10 and Windows Server 2022

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240407171831.zip](https://github.com/OctoPrint/OctoPrint/files/14899595/octoprint-systeminfo-20240407171831.zip)
[install.log](https://github.com/OctoPrint/OctoPrint/files/14899596/install.log)

foosel, 2024-04-08:
Confirmed when installing from a source tarball (which `get.octoprint.org` links to). 

Fix ready for 1.10.0rc4.

What I don't understand is why I see over 500 instances running 1.10.0rc* under Windows... The only explanation that I have is that they installed from PyPi which likely defaulted to the ready made wheel which doesn't have this issue:

```
(testenv) PS C:\Users\gina> pip install OctoPrint==1.10.0rc3
Collecting OctoPrint==1.10.0rc3
  Downloading OctoPrint-1.10.0rc3-py2.py3-none-any.whl.metadata (12 kB)
Collecting OctoPrint-FileCheck>=2021.2.23 (from OctoPrint==1.10.0rc3)
  Using cached OctoPrint_FileCheck-2024.3.27-py2.py3-none-any.whl.metadata (1.7 kB)
Collecting OctoPrint-FirmwareCheck>=2021.10.11 (from OctoPrint==1.10.0rc3)
  Using cached OctoPrint_FirmwareCheck-2021.10.11-py2.py3-none-any.whl.metadata (3.9 kB)
Collecting OctoPrint-PiSupport>=2023.10.10 (from OctoPrint==1.10.0rc3)
  Using cached OctoPrint_PiSupport-2023.10.10-py2.py3-none-any.whl.metadata (2.9 kB)
Collecting argon2-cffi>=23.1.0 (from OctoPrint==1.10.0rc3)
  Using cached argon2_cffi-23.1.0-py3-none-any.whl.metadata (5.2 kB)
Collecting Babel<2.13,>=2.12.1 (from OctoPrint==1.10.0rc3)
  Using cached Babel-2.12.1-py3-none-any.whl.metadata (1.3 kB)
Collecting cachelib<0.11,>=0.10.2 (from OctoPrint==1.10.0rc3)
  Using cached cachelib-0.10.2-py3-none-any.whl.metadata (2.0 kB)
Collecting Click<9,>=8.1.7 (from OctoPrint==1.10.0rc3)
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting colorlog<7,>=6.7.0 (from OctoPrint==1.10.0rc3)
  Using cached colorlog-6.8.2-py3-none-any.whl.metadata (10 kB)
Collecting emoji<3,>=2.10.1 (from OctoPrint==1.10.0rc3)
  Using cached emoji-2.11.0-py2.py3-none-any.whl.metadata (5.3 kB)
Collecting feedparser<7,>=6.0.11 (from OctoPrint==1.10.0rc3)
  Using cached feedparser-6.0.11-py3-none-any.whl.metadata (2.4 kB)
Collecting filetype<2,>=1.2.0 (from OctoPrint==1.10.0rc3)
  Using cached filetype-1.2.0-py2.py3-none-any.whl.metadata (6.5 kB)
Collecting Flask-Assets<3,>=2.1.0 (from OctoPrint==1.10.0rc3)
  Using cached Flask_Assets-2.1.0-py3-none-any.whl.metadata (931 bytes)
Collecting Flask-Babel<4,>=3.1.0 (from OctoPrint==1.10.0rc3)
  Using cached flask_babel-3.1.0-py3-none-any.whl.metadata (2.2 kB)
Collecting Flask-Login<0.7,>=0.6.3 (from OctoPrint==1.10.0rc3)
  Using cached Flask_Login-0.6.3-py3-none-any.whl.metadata (5.8 kB)
Collecting Flask-Limiter<4,>=3.5.0 (from OctoPrint==1.10.0rc3)
  Using cached Flask_Limiter-3.5.1-py3-none-any.whl.metadata (6.1 kB)
Collecting flask<2.3,>=2.2.3 (from OctoPrint==1.10.0rc3)
  Using cached Flask-2.2.5-py3-none-any.whl.metadata (3.9 kB)
Collecting frozendict<3,>=2.4.0 (from OctoPrint==1.10.0rc3)
  Using cached frozendict-2.4.1-cp312-cp312-win_amd64.whl
Collecting future<1,>=0.18.3 (from OctoPrint==1.10.0rc3)
  Using cached future-0.18.3-py3-none-any.whl
Collecting markdown<3.5,>=3.4.4 (from OctoPrint==1.10.0rc3)
  Using cached Markdown-3.4.4-py3-none-any.whl.metadata (6.9 kB)
Collecting netaddr<0.9,>=0.8 (from OctoPrint==1.10.0rc3)
  Using cached netaddr-0.8.0-py2.py3-none-any.whl.metadata (4.9 kB)
Collecting netifaces2<0.1,>=0.0.21 (from OctoPrint==1.10.0rc3)
  Using cached netifaces2-0.0.22-cp37-abi3-win_amd64.whl.metadata (5.3 kB)
Collecting passlib<2,>=1.7.4 (from OctoPrint==1.10.0rc3)
  Using cached passlib-1.7.4-py2.py3-none-any.whl.metadata (1.7 kB)
Collecting pathvalidate<3,>=2.5.2 (from OctoPrint==1.10.0rc3)
  Using cached pathvalidate-2.5.2-py3-none-any.whl.metadata (11 kB)
Collecting pkginfo<2,>=1.9.6 (from OctoPrint==1.10.0rc3)
  Using cached pkginfo-1.10.0-py3-none-any.whl.metadata (11 kB)
Collecting psutil<6,>=5.9.8 (from OctoPrint==1.10.0rc3)
  Using cached psutil-5.9.8-cp37-abi3-win_amd64.whl.metadata (22 kB)
Collecting pydantic==1.10.12 (from OctoPrint==1.10.0rc3)
  Using cached pydantic-1.10.12-py3-none-any.whl.metadata (149 kB)
Collecting pylru<2,>=1.2.1 (from OctoPrint==1.10.0rc3)
  Using cached pylru-1.2.1-py3-none-any.whl.metadata (13 kB)
Collecting pyserial<4,>=3.5 (from OctoPrint==1.10.0rc3)
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting pytz (from OctoPrint==1.10.0rc3)
  Using cached pytz-2024.1-py2.py3-none-any.whl.metadata (22 kB)
Collecting PyYAML<7,>=6.0.1 (from OctoPrint==1.10.0rc3)
  Using cached PyYAML-6.0.1-cp312-cp312-win_amd64.whl.metadata (2.1 kB)
[... 132 lines cut ...]
Using cached idna-3.6-py3-none-any.whl (61 kB)
Using cached ifaddr-0.2.0-py3-none-any.whl (12 kB)
Using cached itsdangerous-2.1.2-py3-none-any.whl (15 kB)
Using cached Jinja2-3.1.3-py3-none-any.whl (133 kB)
Using cached limits-3.10.1-py3-none-any.whl (45 kB)
Using cached MarkupSafe-2.1.5-cp312-cp312-win_amd64.whl (17 kB)
Using cached more_itertools-10.2.0-py3-none-any.whl (57 kB)
Using cached ordered_set-4.1.0-py3-none-any.whl (7.6 kB)
Using cached rich-13.7.1-py3-none-any.whl (240 kB)
Using cached typing_extensions-4.11.0-py3-none-any.whl (34 kB)
Using cached urllib3-2.2.1-py3-none-any.whl (121 kB)
Using cached webassets-2.0-py3-none-any.whl (142 kB)
Using cached argon2_cffi_bindings-21.2.0-cp36-abi3-win_amd64.whl (30 kB)
Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Using cached cffi-1.16.0-cp312-cp312-win_amd64.whl (181 kB)
Using cached Deprecated-1.2.14-py2.py3-none-any.whl (9.6 kB)
Using cached importlib_resources-6.4.0-py3-none-any.whl (38 kB)
Using cached markdown_it_py-3.0.0-py3-none-any.whl (87 kB)
Using cached packaging-24.0-py3-none-any.whl (53 kB)
[... cut ...]  packaging-24.0 passlib-1.7.4 pathvalidate-2.5.2 pkginfo-1.10.0 psutil-5.9.8 pycparser-2.22 pydantic-1.10.12 pygments-2.17.2 pylru-1.2.1 pyserial-3.5 pytz-2024.1 regex-2023.12.25 requests-2.31.0 rich-13.7.1 sarge-0.1.7.post1 semantic-version-2.10.0 sentry-sdk-1.44.1 setuptools-69.2.0 sgmllib3k-1.0.0 tornado-6.2 typing-extensions-4.11.0 unidecode-1.3.8 urllib3-2.2.1 watchdog-2.3.1 webassets-2.0 websocket-client-1.6.1 werkzeug-2.2.3 wheel-0.43.0 wrapt-1.15.0 zeroconf-0.132.0 zipstream-ng-1.7.1
```

MaleNurse, 2024-04-08:
Working now, thank you!

commit 5f4d02d2 (2024-04-08): 🐛 Fix versioning under Windows

setup.py was using a OS-specific relative path for creating the static
version file during build from an sdist ("src/octoprint") which caused
issues under windows.

Closes #4987
closed 2024-04-08 by MaleNurse