OctoPrint/OctoPrint #4988: [RC Feedback] Feedback on the 1.10.0rc4 Release Candidate

foosel, 2024-04-08:
Please provide *general* feedback on your experience with the 1.10.0rc4 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- User and group management functioning as expected.
- Plugin installation functioning as expected.
- Application key management functioning as expected. Authentication workflow with third party clients at your disposal (e.g. slicers) works as it should.
- Backup creation, download and restore functioning as expected

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
###
### ❗👆 IMPORTANT 👆❗

---

### Update test matrix

[image]

*Releases aren't rolled out unless all tests in the update test matrix succeed.*

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

- zeroconf dependency gets built from source instead of pulled from piwheels on OctoPi 0.18, see https://github.com/OctoPrint/OctoPrint/issues/4938#issuecomment-1919869375
  - caused by an outdated pip version, [see this comment](https://github.com/OctoPrint/OctoPrint/issues/4938#issuecomment-1921262160)
  - if affected (OctoPi 0.18.0 development build instead of stable?), make sure to update to `pip>=20.3`
- UI Customizer Plugin: LazeMSS/OctoPrint-UICustomizer#315 - solved with release 0.1.9.91
  - 12-column layout no longer fits
  - camera has additional letterbox

### Issues needing analysis

tigercjn, 2024-04-09:
Dashboard plugin now not displaying data.
[image]

jneilliii, 2024-04-09:
commented on your issue https://github.com/j7126/OctoPrint-Dashboard/issues/403. haven't dug in too deep yet @foosel but looks related to printer profile data binding from the error message in console.

[image]

j7126, 2024-04-09:
> commented on your issue [j7126/OctoPrint-Dashboard#403](https://github.com/j7126/OctoPrint-Dashboard/issues/403). haven't dug in too deep yet but looks related to printer profile data binding from the error message in console.
> 

Looks this is caused by currentProfileData defaulting to null now, instead of an empty profile. changed in https://github.com/OctoPrint/OctoPrint/commit/7123d3361aed570aff3372e05a18ec6809316fb3

j7126, 2024-04-09:
I assume this is intended behavior, as it does make sense, and I will add a null check in the plugin.

tigercjn, 2024-04-09:
It's hard to believe a plugin update now fixes this; it's so quick, thanks to all involved. I love this community.

b-morgan, 2024-04-11:
Updated successfully.

foosel, 2024-04-16:
FYI, I want to give this final RC some more testing over this week and the weekend, and am currently targeting a stable release for next Wednesday (due to a blocker on Tuesday that would stop me from being able to fix anything that might show up after all were I to release on Monday).

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel