OctoPrint/OctoPrint #4989: GCODE Viewer Layer info reports negative number for Filament (Tool 4) with MMU3

MarshMaker3D, 2024-04-09:
### The problem

1.  Configure OctoPrint Printer Profile to use 5 extruders with "Shared nozzle"
2. Load file that uses all 5 "Extruders"
3. Navigate to "GCode Viewer"
4. Check values reported for each "Filament (Tool #):
5. "Filamant Tools" # 0, 1, 2 & 3 report positive numbers
6. "Filamant Tool" # 5 reports a negative number

Occurred with both Stable and RC builds

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.9.3    Python 3.9.2    OctoPi* 1.0.0cam (build 2023.10.09.154319)

### Operating system running OctoPrint

Windows 11 Home Version 23H2 OS Build 22631.3374

### Printer model & used firmware incl. version

Prusa MK3S+ (from kit) with Prusa MMU3 (upgrade from MMU2)

### Browser and version of browser, operating system running browser

Google Chrome Version 123.0.6312.106 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[tornado.log](https://github.com/OctoPrint/OctoPrint/files/14924664/tornado.log)
[plugin_softwareupdate_console.log](https://github.com/OctoPrint/OctoPrint/files/14924665/plugin_softwareupdate_console.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/14924666/octoprint.log)
[octoprint-systeminfo-20240408065149.zip](https://github.com/OctoPrint/OctoPrint/files/14924667/octoprint-systeminfo-20240408065149.zip)
[image]

foosel, 2024-04-10:
Please provide a gcode file with which this can be reproduced, and a screenshot of your printer profile settings in OctoPrint.

MarshMaker3D, 2024-04-12:
Adding GCODE file and Screenshot of printer profile settings
[extruder_spinner_0.2mm_PETG,PETG,PETG,PETG,PETG_MK3SMMU3_1h21m.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/14954408/extruder_spinner_0.2mm_PETG.PETG.PETG.PETG.PETG_MK3SMMU3_1h21m.gcode.txt)
[image]
[image]
[image]
[image]

MarshMaker3D, 2024-04-12:
Please replace Operating system running OctoPrint from "Windows..." to "OctoPi". 
I am a bit of a GitBugNoob...

foosel, 2024-04-15:
From what I can see, this is correctly reflecting the filament usage per tool *per layer*. Note that it says "Layer info" there, so these values apply only to the layer at Z=0.4mm. Looking into the GCODE file, I can see that each tool change involves some heavy retraction:

```
// at this point the active tool is 0 and its total filament usage for the layer sits at 47.67mm

;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : PETG -> PETG
;--------------------
M220 B
M220 S100
; CP TOOLCHANGE UNLOAD
;WIDTH:0.6
G1 X237.384 Y194.674
M900 K0
G1 F2536

G1 X226.816 E0.4895
G1 F2554
G1 X216.176 E0.4929
G1 F2571
G1 X205.463 E0.4962
G1 F2588
G1 X194.679 E0.4996
G1 F2606
G1 X183.821 E0.5029
G1 F2623
G1 X178.884 E0.2287
G1 Y193.834 F7200
G1 F2623
G1 X184.876 E0.2776
G1 F2641
G1 X195.879 E0.5096
G1 F2693
G1 X207.098 E0.5197
G1 F2832
G1 X218.896 E0.5465
G1 F3057
G1 X231.635 E0.5901
G1 F3387
G1 X237.384 E0.2663
G1 Y192.994 F7200
G1 F3387
G1 X229.018 E0.3875
G1 F3804
G1 X213.167 E0.7343
G1 F4221
G1 X195.578 E0.8147
G1 F4517
G1 X178.884 E0.7733
G1 Y192.154 F7200
G1 F4517
G1 X181.010 E0.0985
;WIDTH:0.5

// ... and at this point the total filament usage of T0 is now 54.92mm

// the following lines retract by a combined 50mm, left filament usage is 4.92mm

G1 E-15.0000 F7200 // 39.92
G1 E-24.5000 F1200 // 15.42
G1 E-7.0000 F600 // 8.42
G1 E-3.5000 F360 // 4.92
G4 S0
M104 S230
G1 Y191.314

// extrude and retract 20mm, so 0mm in sum

G1 X237.384 E20.0000 F338
G1 X181.010 E-20.0000 F169
M73 Q2 S79

// retract another 35mm, so now the usage drops to -30.08mm

G1 E-35.0000 F2000
M73 P3 R79
M73 Q3 S79
G1 Y191.484 F2400
M73 P3 R78
G4 S0

// switch to tool 1

; Filament-specific end gcode
T1
```

And depending on how much was extruded by that tool *on that layer* this can result in a negative sum for the tool in question:

[image]

Long story short, the filament usage *per layer* can be negative, depending on the amount of tool changes etc. That of course does not mean that the extrusions on the layer don't happen, it simply means that looking just at the layer, the total sum of extrusions can be negative.

So not a bug, all working fine here.

closed 2024-04-15 by foosel

MarshMaker3D, 2024-04-15:
Thank you for the time, explanation and education.
Apologies for the noise.
I will do better next time.