OctoPrint/OctoPrint #4995: octopi.local Issue

atfam, 2024-04-23:
### The problem

octopi.local   

doesn't work

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

OctoPi

### Browser and version of browser, operating system running browser

chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-logs.zip](https://github.com/OctoPrint/OctoPrint/files/15083640/octoprint-logs.zip)

I use octoeverywhere everywhere to reconnect to it log information if I use the local ip I just it works some reason the octopi.local one doesn't want

jneilliii, 2024-04-23:
This issue tracker here is for OctoPrint bugs and treasure requests. Please reach out on the community forum or discord server for community driven support. First thought is if you are connecting from a Windows machine you need to have the bonjour printing service installed for .local addresses to work. Also, if you changed the hostname while fishing the SD card the address would be hostname.local.

closed 2024-04-23 by jneilliii

atfam, 2024-04-25:
thank you so much