OctoPrint/OctoPrint #4998: Updating to 1.10.0 failed

FrogoYT, 2024-04-27:
### The problem

I tried updating to 1.10.0 in safe mode but I the error message say:

Updating, please wait.

++++++++++++++++++++++++++++++++
Now updating OctoPrint to 1.10.0
++++++++++++++++++++++++++++++++
/usr/bin/python3 -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint/archive/1.10.0.zip --no-cache-dir
Collecting https://github.com/OctoPrint/OctoPrint/archive/1.10.0.zip
Downloading https://github.com/OctoPrint/OctoPrint/archive/1.10.0.zip (6.4 MB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.4/6.4 MB 15.1 MB/s eta 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2021.10.11)
Requirement already satisfied: semantic-version<3,>=2.10.0 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2.10.0)
Collecting blinker<1.7.0,>=1.6.3
Downloading blinker-1.6.3-py3-none-any.whl (13 kB)
Requirement already satisfied: unidecode in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (1.3.8)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2024.3.27)
Requirement already satisfied: flask<2.3,>=2.2.3 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2.2.5)
Requirement already satisfied: pkginfo<2,>=1.9.6 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (1.10.0)
Requirement already satisfied: feedparser<7,>=6.0.11 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (6.0.11)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (0.10.2)
Requirement already satisfied: passlib<2,>=1.7.4 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (1.7.4)
Requirement already satisfied: Flask-Limiter<4,>=3.5.0 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (3.5.1)
Requirement already satisfied: pydantic==1.10.12 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (1.10.12)
Requirement already satisfied: regex in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2022.10.31)
Collecting psutil<6,>=5.9.8
Downloading psutil-5.9.8.tar.gz (503 kB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 503.2/503.2 kB 5.6 MB/s eta 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Requirement already satisfied: requests<3,>=2.31.0 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2.31.0)
Requirement already satisfied: colorlog<7,>=6.7.0 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (6.8.2)
Collecting websocket-client==1.6.1
Downloading websocket_client-1.6.1-py3-none-any.whl (56 kB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 56.9/56.9 kB 37.6 MB/s eta 0:00:00
Requirement already satisfied: wrapt<1.16,>=1.15 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (1.15.0)
Requirement already satisfied: frozendict<3,>=2.4.0 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2.4.1)
Requirement already satisfied: Flask-Login<0.7,>=0.6.3 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (0.6.3)
Collecting markdown<3.5,>=3.4.4
Downloading Markdown-3.4.4-py3-none-any.whl (94 kB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 94.2/94.2 kB 39.5 MB/s eta 0:00:00
Requirement already satisfied: pytz in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2024.1)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2.12.1)
Requirement already satisfied: pathvalidate<3,>=2.5.2 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2.5.2)
Requirement already satisfied: sarge==0.1.7.post1 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (0.1.7.post1)
Requirement already satisfied: class-doc<0.3,>=0.2.6 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (0.2.6)
Requirement already satisfied: future<1,>=0.18.3 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (0.18.3)
Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (0.8.0)
Requirement already satisfied: OctoPrint-PiSupport>=2023.10.10 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2023.10.10)
Requirement already satisfied: wheel in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (0.43.0)
Requirement already satisfied: setuptools in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (65.6.0)
Requirement already satisfied: emoji<3,>=2.10.1 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (2.11.0)
Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (3.1.0)
Collecting netifaces2<0.1,>=0.0.21
Downloading netifaces2-0.0.22-cp37-abi3-musllinux_1_1_aarch64.whl (1.2 MB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 11.5 MB/s eta 0:00:00
Requirement already satisfied: filetype<2,>=1.2.0 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (1.2.0)
Requirement already satisfied: pylru<2,>=1.2.1 in /usr/lib/python3.10/site-packages (from OctoPrint==1.10.0) (1.2.1)
[... 200 lines cut ...]

1.9.3

### Operating system running OctoPrint

OctoPrint 1.9.3 Python 3.10.14

### Printer model & used firmware incl. version

Creality Ender 3 V3 SE, 1.0.6

### Browser and version of browser, operating system running browser

Google Chrome, Windows

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240427123730.zip](https://github.com/OctoPrint/OctoPrint/files/15138183/octoprint-systeminfo-20240427123730.zip)

cp2004, 2024-04-27:
> Operating system running OctoPrint
OctoPrint 1.9.3 Python 3.10.14

Neither of these things are an operating system.

Your OS looks to be Linux, but OctoPrint is installed globally rather than in a virtual environment, so this is not a typical setup. Please provide more details about your setup.

FrogoYT, 2024-04-27:
Oh, I am new to octoprint and GitHub, but yeah, you're right. I'm using an android phone to power octoprint. Sorry I forgot to include that.

foosel, 2024-04-27:
I'm sorry to say that this is an issue with your chosen environment. You are missing a compiler for some native parts of third party dependencies. 

This is not an issue with OctoPrint but an issue with the environment you are trying to set it up under.

You might want to look at octo4a.

closed 2024-04-27 by foosel

cp2004, 2024-04-27:
https://github.com/feelfreelinux/octo4a/releases/tag/1.2.6

Looks like this should have been fixed with Octo4a 1.2.6, released an hour ago.

zaventh, 2024-05-07:
For posterity... I had this issue on Octo4a. It appears the "fix" there is to recreate the entire chroot and reinstall OctoPrint which I wished to avoid. The main issue is 1.10.0 requires `psutil >=5.9.8 <6` while the Octo4a chroot Alpine 3.17 can only provide psutil 5.9.4.

I was able to resolve it by shelling into the chroot and adding build tools and other new dependencies at the dist level:

```shell
apk add build-base gcc python3-dev musl-dev linux-headers py3-zeroconf py3-cffi
apk del ps3-psutil` # removes dist version of psutil at 5.9.4
```
At which point re-running the installer from the OctoPrint UI builds the psutil wheel correctly in the environment.

Simply updating the Octo4a app did not solve the issue for me on an existing installation.