OctoPrint/OctoPrint #4999: Can't use NAS for Octoprint working directories

gbarbehenn, 2024-04-27:
### The problem

Try to assign a directory on an NAS to one of the working directories, say "uploads".  Test will say "The path does not exist and cannot be created."  
The NAS is mapped as a Network drive in Windows 10 Home 64b, and is otherwise completely accessible.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0, not in the lower left hand corner for Firefox on Windows 10 home  64b

### Operating system running OctoPrint

Windows 10 Home 64b WinPE

### Printer model & used firmware incl. version

Elegoo Neptune 3 Pro

### Browser and version of browser, operating system running browser

Firefox 125.0.2, 64b.  Windows 10 Home 64b

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Both the machine running Octoprint (Alfawise Z83) and the machine running the browser (HP Envy I7), have the NAS mapped as a Network drive.  The NAS is mapped to different drive letters on each (Z83--Z:, HP--V:), using either drive letter for the working directory fails.
When I downloaded the Systeminfo Bundle, as required above, it opened the last directory which happened to be on the NAS.  So, it was able to find the NAS, when downloading the System Bundle.
[octoprint-systeminfo-20240427100555.zip](https://github.com/OctoPrint/OctoPrint/files/15139076/octoprint-systeminfo-20240427100555.zip)

cp2004, 2024-04-27:
I have tested on my Windows OctoPrint instance that I can successfully put the uploads folder on an external disk - I don't have a NAS to hand right now, but this should, if the NAS is setup correctly, work the same way.

The error in your log is:

```
2024-04-27 09:48:42,467 - octoprint.server.api - ERROR - Error while trying to create Z:\3D printer\Octoprint\uploads
Traceback (most recent call last):
  File "D:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\site-packages\octoprint\server\api\__init__.py", line 514, in _test_path
    os.makedirs(path)
  File "D:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\os.py", line 215, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "D:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\os.py", line 215, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "D:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\os.py", line 215, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "D:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\os.py", line 225, in makedirs
    mkdir(name, mode)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'Z:\\'
```

Implying that it cannot find a 'Z:' drive. I would be double checking the permissions & the configuration of the drive on the machine that is running OctoPrint - this is the same kind of error I get if I enter the wrong drive number. If the drive is accessible, but the exact folder did not exist, OctoPrint will create it for you.

> Both the machine running Octoprint (Alfawise Z83) and the machine running the browser (HP Envy I7), have the NAS mapped as a Network drive. The NAS is mapped to different drive letters on each (Z83--Z:, HP--V:),

The path should be relative to the *server*, which is on your Alfawise Z83 by the sounds of it. It is not worth trying V:, here, as this is on your client not the server.

> When I downloaded the Systeminfo Bundle, as required above, it opened the last directory which happened to be on the NAS. So, it was able to find the NAS, when downloading the System Bundle.

Could you explain a bit more what you mean by this, what 'it' refers to in this context? My current guess is that when you have downloaded the file to your client, you can see the NAS there. But then this does not make any difference to what the server sees.

foosel, 2024-04-27:
As already mentioned, this is not an OctoPrint issue but rather a problem with how either your NAS is set up (permissions) or drive letter mapping, not OctoPrint. 

Closing this as there's nothing to do here on OctoPrint's end, this is better a bug nor a missing feature. Feel free to move this to the community forums or discord for support.

closed 2024-04-27 by foosel

jneilliii, 2024-04-27:
I bet it has to do with the service running as local system account and those don't have mapped drives. Try with full UNC pathing instead of mapped drive and/or change the service account to run as a specific user that has rights.

[image]