OctoPrint/OctoPrint #5000: Identify server, on an intranet, by name not IP address

gbarbehenn, 2024-04-27:
### Is your feature request related to a problem? Please describe.

When connecting the browser to the service running on another machine, the IP address must be used.  In a DHCP environment, the IP address may change.
I connect to my NAS by the NAS name, not it's IP address.  The DHCP server keeps track of this correspondence, automagically.

### Describe the solution you'd like

That, for intranet access, the server computer name be used instead of the IP address.  This will require interrogating the DHCP server for the name lookup.

### Describe alternatives you've considered

There is only one alternative, use the IP address.  This is fragile in systems that use DHCP, as the IP address might be different on the next usage.

### Additional context

Windows 10 Home 64b.  Firefox 125.0.2 64b.

jneilliii, 2024-04-27:
When accessing from a windows machine depending on your network you can use the hostname to access without issue in most cases. For example I can connect to http://octopi. There is also mDNS in OctoPrint that with the bonjour printing service you can use http://octopi.local for example to access by name. In either case this is not software dependent and is more of a network related topic. This tracker is reserved for bugs and feature requests.

closed 2024-04-27 by jneilliii

gbarbehenn, 2024-04-27:
I did upload the system info bundle, as it was required, and it showed as uploaded.  Reloaded.
I tried the ".local", and modifying the "hosts" file.  However, there seems to be some sort of interaction with DoH (secure DNS over HTTP)?  As such the hosts file is ignored, and Firefox tries to get the IP address from the DNS.
It is possible to map network assets using device name, instead of IP address.  But, perhaps as you say, this is up to the browser, not the application.  Shame though, as DHCP address changes can create sudden unexpected problems when using IP addresses.
I have never had a need for Bonjour, or Bonjour printing services, and so don't have it running on my PC.
[octoprint-systeminfo-20240427100555.zip](https://github.com/OctoPrint/OctoPrint/files/15140079/octoprint-systeminfo-20240427100555.zip)

cp2004, 2024-04-27:
OctoPrint doesn't know or care if you access it with the hostname or IP address. Windows is notoriously bad at resolving hostnames, which is why installing Bonjour helps.

jneilliii, 2024-04-27:
> DHCP address changes can create sudden unexpected problems when using IP addresses.

it is possible to set DHCP reservations to keep the IP the same based on MAC address, that's what I do on my router.

gbarbehenn, 2024-04-28:
Some of my network resources use fixed IP addresses, some use MAC address filtering, and some get their address (more or less) randomly at ARP.  I say more or less randomnly because most DHCP servers allow the assignment of a "preferred" IP address, but I don't know what that means.  This wasn't a conscious design, but evolved over time, as has been stable for several years now.    But, in any event fixed IP addresses aren't discoverable, and MAC address filtering would need to be obtained from the DHCP server.  And the DoH implementation only seems to know about DNSes outside the intranet?
In theory fixed IP addresses should be available in the Hosts file, but apparently Firefox ignores the Hosts file for single word URLs when DoH is enabled?

yasavvym8, 2024-04-29:
I can't remember if this is what I did to solve this problem in do, but if you ssh into octoprint with pi@octoprint.local (Or whatever your username is @ Whatever your server is called), this might allow you to log in on your browser with just octopi.local on windows. I can't remember exactly what I did But for some reason this is coming to mind. I would also try to SSH with whenever the current IP address is also For good measure