OctoPrint/OctoPrint #5007: [Request] make timezone for achievements more prominent

jonyo, 2024-05-04:
### Is your feature request related to a problem? Please describe.

When I first saw achievements, the time based ones were wrong. I got TGIF on a Thursday, "night owl" in the afternoon, etc.

There was no indication in the achievement that it was based on specific timezone. I found the timezone setting eventually, but for the first week or so I didn't look that close and just thought it was a new feature that was a little buggy. And I did not see any CTA to set the timezone (that I can remember).

### Describe the solution you'd like

2 possible things:

1. Call to action to set the timezone in the UI.  Maybe as a notification or something. 
2. Ability to reset just the time/date based achievements - it bugged me I had a bunch for the wrong thing so I cleared the plugin data, and that cleared all my achievements not just time/date ones. So it would be cool to have some way in the UI to reset all time related, or maybe "reset this achievement" on individual ones.

### Describe alternatives you've considered

Another idea, expanding on the first one more: Could have a "setting the timezone" achievement!

Maybe all time related achievements are "locked" until you specifically set the timezone and confirm it.

Maybe a "use local timezone" button that would use JS to get the localized timezone (though I know from experience that can be tricky... maybe it is easier these days though I haven't tried in a while).

foosel, 2024-05-07:
I've done this:

- timezone information, "time based" indicator and link to configuration in achievements overview
  [image]
- ability to reset time based or all achievements from the settings
  [image]
- offer to reset all time based achievements on change of the configured timezone
  [image]

Using the client side timezone wouldn't work - there might not be a client connected at the time the achievement is unlocked, so I have to rely on the server time. I tried squeezing a link to the config into time based achievement unlock notifications, but everything I tried to achieve that just looked horrible.

Will be in 1.10.1

jonyo, 2024-05-07:
The screenshot look great, I like that chino of solutions. The option to reset all from the settings page is good touch as well, not everyone would think to try clearing the plug-in data on a different part of the ui.

:tada:

commit af863e67 (2024-05-07): 🚸 achievements: Clarify timezone config requirement

- show currently configured (or used) timezone/time
- link to achievements config w/ timezone config from about dialog
  (if having settings permission)
- add a symbol to all time based achievements on the UI
- allow to reset only those (or all)
- offer reset of time based achievements on change of configured
  timezone

Closes #5007
closed 2024-05-14 by foosel