OctoPrint/OctoPrint #5008: Metadata pre-processing doesn't work with gcode files containing whitespaces.

ofoxus, 2024-05-05:
### The problem

When uploading a gcode file containing a whitespace, the metadata processor won't be able to execute the octoprint analysis code. E.g. I tried printing a file called `CFFFP_medium paint palette.gcode` (from Cura):
`/usr/local/bin/python -m octoprint analysis gcode --speed-x=15000 --speed-y=15000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /octoprint/octoprint/uploads/CFFFP_medium paint palette.gcode`
This fails with the following error:
```
Usage: octoprint analysis gcode [OPTIONS] PATH
Try 'octoprint analysis gcode --help' for help.

Error: Got unexpected extra arguments (paint palette.gcode)
```

Expected result (I added the missing quotes manually):
```
# /usr/local/bin/python -m octoprint analysis gcode --speed-x=15000 --speed-y=15000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 "/octoprint/octoprint/uploads/CFFFP_medium paint palette.gcode"
DONE:1.0590811959991697s
RESULTS:
dimensions:
  depth: 168.88
  height: 2.2
  width: 183.256
extrusion_length:
- 5866.855039999981
extrusion_volume:
- 0.0
printing_area:
  maxX: 183.256
  maxY: 163.88
  maxZ: 2.2
  minX: 0.0
  minY: -5.0
  minZ: 0.0
total_time: 117.87162016083214
travel_area:
  maxX: 200.0
  maxY: 220.0
  maxZ: 10.0
  minX: 0.0
  minY: -5.0
  minZ: 0.0
travel_dimensions:
  depth: 225.0
  height: 10.0
  width: 200.0
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0

### Operating system running OctoPrint

Linux - Raspberry Pi OS Lite (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Systeminfo Bundle:
[octoprint-systeminfo-20240505121238.zip](https://github.com/OctoPrint/OctoPrint/files/15212901/octoprint-systeminfo-20240505121238.zip)

GCODE file used:
[CFFFP_medium paint palette.zip](https://github.com/OctoPrint/OctoPrint/files/15212911/CFFFP_medium.paint.palette.zip)

I already intend on creating a pull request for the problem, as it is relatively easy to fix.

foosel, 2024-05-06:
Thanks! Weird that this didn't show up so far... In any case, I have merged your fix (:+1:) and also backported it to `staging/bugfix`, so this will be fixed in 1.10.1.

ofoxus, 2024-05-07:
Update on this issue. I left my octoprint running for 2 days and the strangest things started to happen with my octoprint installation. My fix did work for me then, but it started  acting weird today. My monkeypatched version started faulting in another area: the quotes passed to sarge weren't stripped in the call to `octoprint/cli/analysis.py` resulting in that module not finding the file anymore. I decided to wipe my installation and reinstall it again and now the old version without the quotes works too!

<details>

<summary> .metadata.json now </summary>

```
cat /octoprint/octoprint/uploads/.metadata.json
{
  "test2.gcode": {
    "hash": "8ff8f92aa165d8c7c3f52829db74d0810d86471e",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/test2.png?20240507190003",
    "thumbnail_src": "prusaslicerthumbnails",
    "analysis": {
      "printingArea": {
        "maxX": 162.985,
        "maxY": 184.212,
        "maxZ": 15.0,
        "minX": 0.0,
        "minY": -5.0,
        "minZ": 0.0
      },
      "dimensions": {
        "depth": 189.212,
        "height": 15.0,
        "width": 162.985
      },
      "travelArea": {
        "maxX": 200.0,
        "maxY": 220.0,
        "maxZ": 20.0,
        "minX": 0.0,
        "minY": -5.0,
        "minZ": 0.0
      },
      "travelDimensions": {
        "depth": 225.0,
        "height": 20.0,
        "width": 200.0
      },
      "estimatedPrintTime": 12142.82217389923,
      "filament": {
        "tool0": {
          "length": 9961.599939999678,
          "volume": 0.0
        }
      }
    },
    "dashboard": {
      "layer_move_array": "[8, 5921, 3672, 3754, 3856, 3890, 3914, 4328, 4685, 4206, 4386, 2217, 2229, 2220, 2235, 2250, 643, 656, 651, 647, 612, 639, 644, 665, 661, 869, 880, 890, 799, 801, 836, 761, 851, 865, 811, 591, 619, 612, 611, 611, 611, 612, 616, 635, 636, 647, 645, 655, 642, 645, 650, 507, 516, 499, 583, 588, 596, 575, 578, 588, 587, 607, 619, 667, 668, 461, 452, 413, 414, 411, 262, 259, 269, 266, 264, 261]",
      "filament_change_array": "[]"
    },
    "history": [
      {
        "timestamp": 1715108778.0424905,
        "success": false,
        "printerProfile": "_default"
      }
    ],
    "statistics": {
      "averagePrintTime": {},
      "lastPrintTime": {}
    }
  },
  "test2 - Copy.gcode": {
    "hash": "8ff8f92aa165d8c7c3f52829db74d0810d86471e",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/test2%20-%20Copy.png?20240507190018",
    "thumbnail_src": "prusaslicerthumbnails",
    "analysis": {
      "printingArea": {
        "maxX": 162.985,
        "maxY": 184.212,
        "maxZ": 15.0,
        "minX": 0.0,
        "minY": -5.0,
        "minZ": 0.0
      },
      "dimensions": {
        "depth": 189.212,
        "height": 15.0,
        "width": 162.985
      },
      "travelArea": {
        "maxX": 200.0,
        "maxY": 220.0,
        "maxZ": 20.0,
        "minX": 0.0,
        "minY": -5.0,
        "minZ": 0.0
      },
      "travelDimensions": {
        "depth": 225.0,
        "height": 20.0,
        "width": 200.0
      },
      "estimatedPrintTime": 12142.82217389923,
      "filament": {
        "tool0": {
          "length": 9961.599939999678,
          "volume": 0.0
        }
      }
    },
    "dashboard": {
      "layer_move_array": "[8, 5921, 3672, 3754, 3856, 3890, 3914, 4328, 4685, 4206, 4386, 2217, 2229, 2220, 2235, 2250, 643, 656, 651, 647, 612, 639, 644, 665, 661, 869, 880, 890, 799, 801, 836, 761, 851, 865, 811, 591, 619, 612, 611, 611, 611, 612, 616, 635, 636, 647, 645, 655, 642, 645, 650, 507, 516, 499, 583, 588, 596, 575, 578, 588, 587, 607, 619, 667, 668, 461, 452, 413, 414, 411, 262, 259, 269, 266, 264, 261]",
      "filament_change_array": "[]"
    }
  },
  "CFFFP_Art Palette.gcode": {
    "hash": "5dc1acc6868494dc97853b3d4c236412fdedac4a",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/CFFFP_Art%20Palette.png?20240507190624",
    "thumbnail_src": "prusaslicerthumbnails",
    "analysis": {
      "printingArea": {
        "maxX": 212.836,
        "maxY": 187.46,
        "maxZ": 4.8,
        "minX": 0.0,
        "minY": -5.0,
        "minZ": 0.0
      },
      "dimensions": {
        "depth": 192.46,
        "height": 4.8,
        "width": 212.836
      },
      "travelArea": {
        "maxX": 212.836,
        "maxY": 220.0,
        "maxZ": 10.0,
        "minX": 0.0,
        "minY": -5.0,
        "minZ": 0.0
      },
      "travelDimensions": {
        "depth": 225.0,
        "height": 10.0,
        "width": 212.836
      },
      "estimatedPrintTime": 27858.101268077822,
      "filament": {
        "tool0": {
          "length": 24140.12336000039,
          "volume": 0.0
        }
      }
    },
    "dashboard": {
      "layer_move_array": "[8, 2348, 2371, 2339, 2374, 2343, 2376, 4700, 8224, 7691, 7883, 8144, 8295, 5878, 5869, 5879, 5008, 6029, 6061, 6790, 6832, 6803, 6862, 7568, 7001]",
      "filament_change_array": "[]"
    }
  }
}
```

</details>

I am really confused by this, as the script is not runnable from the console by entering
`/usr/local/bin/python -m octoprint analysis gcode --speed-x=15000 --speed-y=15000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /octoprint/octoprint/uploads/test whitespace.gcode` and invoking it from a python script shouldn't differ in behavior.
I wouldn't mind replacing the command string passed to sarge by an arg array call to `sarge.Command`, just to be on the safe side.
As it only happened to me though, I would recommend reverting my PR for now and I will keep you updated on this if you don't mind.

foosel, 2024-05-13:
I also missed during review that this already *was* passing the command as an array. Note how it doesn't use `' '.join(command)` for the actual `sarge.run` invocation, there it uses `command`, which is a full array.

I have no idea why it originally failed for you, but from the code it already did the right thing (and I have no clue why I missed that during the PR review - the past few weeks were a bit stressful however, so I'll chalk it up to that).

As it looks like there never was an actual bug here (at least not as far as we both can currently see), I'm closing this, but comments are still possible should you have any updates!

closed 2024-05-13 by foosel

ofoxus, 2024-05-15:
Now that you say it, you are totally right and I am a complete fool. It is already being passed to `sarge.run` as an array. I am truly sorry!

I will keep a tighter watch on my installation the next time around, should I re-encounter it. It's just so strange and I wish I could roll back to before the re-installation to track down the real culprit. First the .metadata.json wasn't filled and afterwards the empty default set was always used.