OctoPrint/OctoPrint #5017: Achievements plugin: unsupported format character 'Z' (0x5a) at index 93

armybean, 2024-05-21:
### The problem

When I open the achievements plugin, I get this error:

`There was an error with the template dialogs/about.jinja2 at line number 48: unsupported format character 'Z' (0x5a) at index 93`

This appears since I updated to 1.10.1 - in 1.10.0 it was possible to open the achievements page as normal.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.10.1

### Operating system running OctoPrint

OctoPi* 1.0.0cam (build 2023.10.09.154319)

### Printer model & used firmware incl. version

Prusa MK3S+ MMU3 version 3.13.2

### Browser and version of browser, operating system running browser

Brave 1.65.132 on macOS Monterey 12.7.4

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240521110043.zip](https://github.com/OctoPrint/OctoPrint/files/15387395/octoprint-systeminfo-20240521110043.zip)
[image]

foosel, 2024-05-21:
This seems to be an issue with german translation, does it stop to reproduce when you switch it to english? (Can be done in the User Settings or globally in the Appearance Settings)

foosel, 2024-05-21:
Just confirmed that this only happens with the (bundled) german translation, so it must be something in there.

armybean, 2024-05-21:
Maybe here: 
OctoPrint/translations/de/LC_MESSAGES/messages.po (beginning line 300)

```
#: src/octoprint/plugins/achievements/templates/achievements_about_achievements.jinja2:6
#, python-format
msgid ""
"Achievements with a %(clock)s are time based and rely on proper "
"%(startlink)stimezone configuration%(endlink)s (currently: %(timezone)s, "
"%(time)s)."
msgstr ""
"Achievements mit einem %(clock)s sind zeitbasiert und hängen von einer "
"korrekten %(startlink)Zeitzonen-Konfiguration%(endlink) ab (aktuell: "
"%(timezone)s, %(time)s)."
```

In the English part there is %(startlink)s and %(endlink)s where in German translation there is the "s" missing.

commit 522e4c8a (2024-05-21): 🐛 Fix translation error leading to template render error

Closes #5017
closed 2024-06-18 by foosel