OctoPrint/OctoPrint #5018: [Request] Allow API for setting "Current_Extruder" for Multi-Toolhead Printers

ccatlett1984, 2024-05-22:
### Is your feature request related to a problem? Please describe.

If you have multiple printers, there is no way to send a temp command for the current tool:
M104 T{current_extruder} S140

### Describe the solution you'd like

I've taken a crack at editing standard.py, but must have a syntax error. Not a python person ;)

Adding this additional endpoint, would allow for setting hotend temp of the loaded tool, without having to specify the tool (and in the case of OctoDash being hard coded to tool0 currently.)

```py

    def set_temperature(self, heater, value, *args, **kwargs):
        if not PrinterInterface.valid_heater_regex.match(heater):
            raise ValueError(
                'heater must match "tool([0-9]+|Current)", "bed" or "chamber": {heater}'.format(
                    heater=heater
                )
            )

        if not isinstance(value, (int, float)) or value < 0:
            raise ValueError(f"value must be a valid number >= 0: {value}")

        tags = kwargs.get("tags", set()) | {"trigger:printer.set_temperature"}

        if heater.startswith("tool"):
            printer_profile = self._printerProfileManager.get_current_or_default()
            extruder_count = printer_profile["extruder"]["count"]
            shared_nozzle = printer_profile["extruder"]["sharedNozzle"]
            if extruder_count > 1 and not shared_nozzle:
                if heater[len("tool") :] == "Current":
                    self.commands(f"M104 T{"current_extruder"} S{value}", tags=tags)
                else:
                    toolNum = int(heater[len("tool") :])
                self.commands(f"M104 T{toolNum} S{value}", tags=tags)
            else:
                self.commands(f"M104 S{value}", tags=tags)
```

### Describe alternatives you've considered

Since there does not be a method to query a Marlin (or other OS) printer for which toolhead is active, I don't see another way for controlling the "current" toolhead temp.

morphias2004, 2024-05-22:
+ 1

morphias2004, 2024-05-23:
A friend of @ccatlett1984 and I (Zura - thanks dude!) helped with the syntax errors, so OctoPrint now boots without issue.

Attached is the working standard.py file.
[standard.py.txt](https://github.com/OctoPrint/OctoPrint/files/15424949/standard.py.txt)

ccatlett1984, 2024-05-25:
https://github.com/OctoPrint/OctoPrint/pull/5022

foosel, 2024-05-27:
Quoting myself here:

> I've changed the implementation a bit.
> 
> First of all you forgot to adjust the regexes used for validating tool names. Then, I really wasn't happy with something like `toolCurrent_Extruder` - instead it's now just `tool` without a number, and it is only supported for setting a temperature, which is the only place where it even makes sense (so I also added another validation regex to ensure that on e.g. setting an offset or selecting a tool it would not be supported). 
> 
> Then I also adjusted the actual REST API endpoint, which also contains a sanity check which got forgotten, fixed a bug in the virtual printer plugin (`M104 S120` would set the temperature of the first tool, instead of the current one) and finally updated the docs as well.
> 
> So, long story short, something like this works now:
> 
> ```
> curl --header "Authorization: Bearer ..." http://example.com/api/printer/tool --json '{"command": "target", "targets": {"tool": 120}}'
> ```
> 
> and is also documented correctly.

Will be in 1.11.0 :+1: Docs are already updated here: https://docs.octoprint.org/en/maintenance/api/printer.html#issue-a-tool-command

ccatlett1984, 2024-05-27:
Awesome, now on to getting OctoDash sorted.

closed 2024-05-27 by ccatlett1984