OctoPrint/OctoPrint #5019: 1.10.0 breaks CLI dev plugin:new

jneilliii, 2024-05-22:
### The problem

https://discord.com/channels/704958479194128507/708230829050036236/1241028558676299917

When trying to run `octoprint dev plugin:new` command is not recognized.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

CLI command related

### Version of OctoPrint

1.10.1

### Operating system running OctoPrint

Windows, OctoPi

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

n/a

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

1.10.1 result of `octoprint dev --help`
[image]

1.9.3 result of `octoprint dev --help` prior to upgrade
[image]

strange thing is downgrading back to OctoPrint 1.9.3 doesn't bring it back. 

[octoprint-systeminfo-20240522005551.zip](https://github.com/OctoPrint/OctoPrint/files/15398151/octoprint-systeminfo-20240522005551.zip)

foosel, 2024-05-22:
Uhm...

[image]

Now that's going to get interesting to figure out.

Fresh install into a new venv:

[image]

This reeks of a third party dependency update breaking something here in a subtle way, which would also explain why going back to 1.9.3 doesn't fix it.

It doesn't look like it's the python version, as you reproduced it under 3.9 (based on your system info bundle) and I couldn't reproduce it under an existing venv created for 3.11 but could reproduce in a fresh venv under 3.12.

Can you please do a `pip freeze > pip-freeze.txt` and attach the resulting txt file?

foosel, 2024-05-22:
Never mind, found it. It's something about cookiecutter 2.6.0, released on Feb 21st. If I manually downgrade to cookiecutter 2.5 in the new venv, the command pops up.

foosel, 2024-05-22:
Fix/workaround ready for 1.10.2.

An import location changed in cookiecutter 2.6.0 (or rather, something was no longer being imported into `cookiecutter.prompt` which previously was), so a sanity check by OctoPrint fell on its face.

commit 56bb9165 (2024-05-22): 🐛 Fix plugin:new cli command

StrictEnvironment can no longer be imported from
cookiecutter.prompt in cookiecutter 2.6+, so let's import it from
the right location, and also fail properly if the import fails.

Closes #5019
closed 2024-06-18 by foosel