OctoPrint/OctoPrint #5020: after a svg version of icon to make a better custombrand icons version

rchiileea, 2024-05-22:
### Is your feature request related to a problem? Please describe.

Hi there, I am reaching out to see if I can grab the svg icon to make a solid version of the icon for custom brand icons, we do have a user submited one but it is jerky and blocky, I will convert it to a flat icon for the repo (I have messaged you directly on facebook so your original color svg of icon does not get into the public domain.

### Describe the solution you'd like

svg icon version

### Describe alternatives you've considered

non applical

cp2004, 2024-05-22:
There are higher resolution versions of the icons available here

https://octoprint.org/trademark-rules/

https://octoprint.org/trademark-rules/logo_png_kit.zip

And an SVG of the tentacle exists in the repo already
 https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/static/img/mask-theme.svg

rchiileea, 2024-05-22:
> There are higher resolution versions of the icons available here
>
> https://octoprint.org/trademark-rules/
>
> https://octoprint.org/trademark-rules/logo_png_kit.zip
>
> And an SVG of the tentacle exists in [...]

Thanks Charlie, but I was looking for a svg of the actual main logo to convert to a better flat one that the user submited one we have.
[image]

I mean I can draw the icon from the high res png anyway, but thought you guys and gals maybe able to help out another hacs repo.

foosel, 2024-05-22:
> we do have a user submited one but it is jerky and blocky

Uhm, who is "we"? 😅 Or in other words, what is this for?

In general I'd prefer to see the tentacle icon used for application icons, as that is way better suited for square and tiny rendering.

> I have messaged you directly on facebook so your original color svg of icon does not get into the public domain

An email is better. While I still do have a Facebook account, I don't use it apart for maintaining the OctoPrint page on there (and I'm more and more leaning towards giving that the axe too) and messages sent via Facebook messenger are not actively monitored. Got way too much crap on there so I even disabled all notifications. You can always reach me under gina -at- octoprint -dot- org.

foosel, 2024-05-22:
Yeah, then *definitely* please use the tentacle there, the large octopus logo is really utterly unsuited to act as an icon. Even in a higher resolution, the result will turn out to just look like crap at the small icon size (believe me, I tried - I have a Home Assistant install myself ^^). Which is also the reason why I use the tentacle icon as a favicon rather than the octopus. 

Here's the relevant SVG files for that:

[tentacle-svg.zip](https://github.com/OctoPrint/OctoPrint/files/15400952/tentacle-svg.zip)

rchiileea, 2024-05-22:
I will add the tenticle as mobile option, I just quickly through this one together too
[image]

foosel, 2024-05-22:
That looks great! But please really only use that where it has enough space to shine 😅 

I have had a bunch of situation where I fell over the logo in situations where the tentacle would have been a much much better fit, and every time that happens it makes me sad as it really does make things look worse than they could.

foosel, 2024-05-29:
Closing since I believe this is solved now.

closed 2024-05-29 by foosel