OctoPrint/OctoPrint #5021: [BUG] Hide Succesful printed files hides folders also

notoriouscarl, 2024-05-23:
### The problem

Hide successful printed files hides folders also. while enabled, it will only show folders that have NO successful printed files. even if the folder has un-printed or unsuccessful prints in the folder the folder is hidden. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.1

### Operating system running OctoPrint

OctoPi 0.17.0

### Printer model & used firmware incl. version

Ender 3, custom firmware

### Browser and version of browser, operating system running browser

Firefox 126.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2024-05-27:
Confirmed. Fix is ready for 1.10.2.

The logic for folders was broken in general as it would hide folders that have only one successful print on ANY file in them, so the logic has now changed:

- folders will be shown as long as they have at least one file in them that has not been printed successfully yet (recursively!)
- folders will also be shown if they contain the currently selected file, regardless of the amount of successful prints

notoriouscarl, 2024-05-27:
thanks for looking into it.

commit 81114bfd (2024-05-27): 🐛 Fix "printed files" filter

It was also hiding folders.

The logic for folders was broken anyhow as it would hide
folders that have only one successful print on ANY file in
them, so the logic has now changed:

- folders will be shown as long as they have at least one
  file in them that has not been printed successfully yet
  (recursively!)
- folders will also be shown if they contain the currently
  selected file, regardless of the amount of successful
  prints

Closes #5021
closed 2024-06-18 by foosel