OctoPrint/OctoPrint #5024: Achievement Plugin disabled, but achievements still occurs

Lars-O-Knudsen, 2024-05-28:
### The problem

I have disabled the achievement plugin
And also restarted server
Still an achievement pop up.
The achievement feature is really not my thing - I strongly oppose that this plugin can not be un installed, really??? 
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.1

### Operating system running OctoPrint

OctoPi

### Browser and version of browser, operating system running browser

Brave v1.66.113, Chromium 125.0.6422.76

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240528120748.zip](https://github.com/OctoPrint/OctoPrint/files/15467570/octoprint-systeminfo-20240528120748.zip)

foosel, 2024-05-28:
Your screenshot shows that it in fact is still active as the server has *not* yet been restarted. Your log also indicates that you first only cleaned up its data (which does *not* disable it, it really only resets its data), then you restarted, the plugin was still enabled, then you installed another plugin, triggering an achievement and apparently also set the Achievements plugin to disabled, then you restarted *again* and in your latest restart it actually *is* disabled, just as it should be. 

I just also tested this...
- Disabled the achievement plugin (note the little reload icon that means it's not yet disabled and still requires a restart, note also that it's NOT greyed out): [image]
- Restarted, confirmed it was greyed out: [image]
- Triggered an achievement condition, did not get an achievement
- Reenabled it, restarted, verified it was active again: [image]
- Triggered the condition again, got an achievement
- Just to make sure this wasn't some weird issue caused by "cleanup plugin data", I then cleaned up the plugin data and also disabled the plugin, restarted, and it was disabled just fine

So, I'm sorry to say, I cannot reproduce this at all. I also know for a fact that plenty of people out there have disabled it and other bundled plugins successfully, and disabling a plugin will make it so OctoPrint won't even try to load it at all, so it can't possibly run any code in that case. It would also be something that goes way beyond this plugin in particular if disabling wasn't working out at all.

My current guess as to what happened here is thus that you cleaned up the data, thought that also disabled it, restarted, installed a new plugin, got an achievement, got frustrated, clicked on disable, and since your last restart this should now actually be disabled and you should be set.

Should for some reason something have not worked here after all, feel free to reopen this and we can look further into it, but based on the above tests, this issue is something rooted in a misunderstanding of the "cleanup data" button.

closed 2024-05-28 by foosel

Lars-O-Knudsen, 2024-05-30:
Hi Gina,
Thanks for your comprehensive answer.

I feel quite convinced I did (try to) disable it before I cleared the data. But the pi is in the far room of the basement and wifi conn can be quite bad there. Maybe I got it clicked twice because it can be very late in response.
Sorry for inconvenience.