OctoPrint/OctoPrint #5027: [Achievements] "The Adventurer" achievement is not unlocked by installing a plugin from an URL

ghudiczius, 2024-05-31:
### The problem

It is just a minor nit and not even sure whether it is intentional or not, but after a clean OctoPrint install, if the first plugin is added via a URL, the achievement _"The adventurer"_ is not unlocked, only the achievement _"The Tinkerer"_ is.

[image]

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Bug is related to the Achievements plugin

### Version of OctoPrint

1.10.1

### Operating system running OctoPrint

Raspberry Pi OS (64 bit, bookworm, lite)

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

ghudiczius, 2024-05-31:
[octoprint-systeminfo-20240531235806.zip](https://github.com/user-attachments/files/15519354/octoprint-systeminfo-20240531235806.zip)

ghudiczius, 2024-05-31:
[octoprint-systeminfo-20240601010416.zip](https://github.com/user-attachments/files/15519369/octoprint-systeminfo-20240601010416.zip)

foosel, 2024-06-04:
That is actually intentional, though now you've made me question whether it make sense this way 😅

The way the achievement is worded, it definitely doesn't... Hm.

foosel, 2024-06-05:
I'll classify that as a description error in the achievement. It should specify that it relates to a plugin installed from the repo. That's now fixed, will roll out as part of 1.10.2.

commit e4512750 (2024-06-05): 💬 achievements: Fix description of Adventurer plugin

Closes #5027
closed 2024-06-18 by foosel