OctoPrint/OctoPrint #5030: Printer is freezing while printing.

Maniekgie, 2024-06-15:
### The problem

Printer is freezing while printing for some seconds and its making filaments dots on model. 
[uchwyt okularow.gcode.txt](https://github.com/user-attachments/files/15845004/uchwyt.okularow.gcode.txt)
[image]
Left is from octoprint and right is directly from sd card (offline) 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.1

### Operating system running OctoPrint

Linux (android) 

### Printer model & used firmware incl. version

Ender 3 pro 

### Browser and version of browser, operating system running browser

All 

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]
Left is directly from pc via octoprint and right is directly from sd card (offline version) 
[uchwyt okularow.gcode.txt](https://github.com/user-attachments/files/15844980/uchwyt.okularow.gcode.txt)

Maniekgie, 2024-06-15:
@foosel can you help?
I changed usb cable to better and same
Connection is not breaking i think but better to See in logs

jneilliii, 2024-06-16:
That is very disrespectful to ping someone on an issue that you haven't given more than a day to respond. Gina (foosel) only works on this project during the week, and specifically on certain tasks that are scheduled. What you describe is not actually a bug with OctoPinrt but more than likely a problem with your printer's firmware buffer configuration not being large enough to store a bunch of movements being sent from USB. This would typically require compiling and reflashing your printer's main board's firmware with the appropriate changes. I would recommend you get in touch on the community forum or Discord server for further assistance.

Maniekgie, 2024-06-16:
Im new user, sorry. I didnt know to See this topic by author.
Its not problem of frimware (buffer) of my printer because When im printing by cura from my Windows by usb it works very well and printing is no freezing.
I think something is wrong with my octoprint.

foosel, 2024-06-17:
Based on your log, OctoPrint is constantly running into communication timeouts with your printer, meaning something is up with your connection between whatever you are running OctoPrint on (it looks like a custom install) and your printer. That explains the freezes. Could be a wonky USB cable, broken firmware, several other factors. All of that is [covered in the FAQ](https://community.octoprint.org/docs?topic=227). 

Now, for questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)