OctoPrint/OctoPrint #5033: API Octopi (running on Pi Zero 2W in my case) reports wrong volume value

PitHerm, 2024-06-23:
### The problem

1. Starting a printjob on Octopi

2. Accessing Octopi running on Pi Zero 2 W using API with Firefox sending job information request
   http://MY_OCTOPI_IP/api/job?apikey=MY_API_KEY
3. Result, Volume is always 0 :
    <html><body>

job |  
-- | --
averagePrintTime | 2304.1957305034994
estimatedPrintTime | 1345.9887080730473
filament |  
tool0 |  
length | 1166.406040000003
volume | 0
file |  
date | 1719117807
display | "Test_your_3D_printer_v2_0.48mm_V1.gcode"
name | "Test_your_3D_printer_v2_0.48mm_V1.gcode"
origin | "local"
path | "Test_your_3D_printer_v2_0.48mm_V1.gcode"
size | 1226644
lastPrintTime | 2324.9526363589994
user | "pi"
progress |  
completion | 100
filepos | 1226644
printTime | 2325
printTimeLeft | 0
printTimeLeftOrigin | null
state | "Operational"

</body>
</html>

5. depending on you docs: https://docs.octoprint.org/en/master/api/job.html#retrieve-information-about-the-current-job
    it should be reported (from docs):
    {
  "job": {
    "file": {
      "name": "whistle_v2.gcode",
      "origin": "local",
      "size": 1468987,
      "date": 1378847754
    },
    "estimatedPrintTime": 8811,
    "filament": {
      "tool0": {
        "length": 810,
        "volume": 5.36
      }
    }
  },

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

 Connection Serial Port Baudrate Printer Profile Save connection settings Auto-connect on server startup State State: Operational Resend ratio: 0 / 87.9K (0%) File: Test_your_3D_printer_v2_0.48mm_V1.gcode  Uploaded: 2024-06-23 06:43:27 User: pi Filament (Tool 0): 1.17m Approx. Total Print Time: 40 minutes Print Time: 00:38:45 Print Time Left: - Printed: 1.2MB / 1.2MB Current Height: 17.2 / 7.0mm Layer: 34 / 34 100% Files 9xFelix_Key-Chain_V2_text.gcode Updated: 18 hours ago Size: 15.0MB Layers: 29 Test_your_3D_printer_v2_0.48mm_V1.gcode Updated: 2 hours ago Size: 1.2MB Layers: 34 Free: 4.3GB / Total: 7.0GB Create folder... Upload Upload to SD Hint: You can also drag and drop files on this page to upload them. Printer Notifications There are currently no notifications from your printer.      Temperature     Control     GCode Viewer     Terminal     Resource Monitor  50°C 100°C 150°C 200°C 250°C 300°C now -5min -10min -15min -20min -25min 	Actual T: 27.7°C	 	Target T: off 	Actual Bed: 26.8°C	 	Target Bed: off 	Actual 	Target 	Offset Tool 	27.7°C 	 °C 	 0 °C Bed 	26.8°C 	 °C 	 0 °C      OctoPrint 1.10.2 Python 3.9.2 OctoPi* 1.0.0 (build 2024.05.14.100113)

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender S1 Pro

### Browser and version of browser, operating system running browser

Firfox 127.0.1 64 bit, OS: Windows 10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240623101944.zip](https://github.com/user-attachments/files/15943840/octoprint-systeminfo-20240623101944.zip)

cp2004, 2024-06-23:
Make sure you have set the filament diameter correctly in your printer profile. Also, please share a gcode file that we can reproduce this with.

PitHerm, 2024-06-23:
@cp2004 sorry but i don't spot a filament setting within the printer profile ?
[image]
[image]
[image]
[image]

PitHerm, 2024-06-23:
@cp2004 sorry about my mistake :-( and THANKS for point me the right direction. the Ultimaker and the Prusa slicer didn't include an M200 D1.75 command . After including it into the printer profiles there everything is fine . Thanks again.

closed 2024-06-23 by PitHerm