OctoPrint/OctoPrint #5034: Processor hitting 100%, fixes after changing feedrate.

andrebstv, 2024-06-26:
### The problem

It's quite an evasive bug. If I do print this file it prints the 1st layer flawlessly. After that it frequently (like at least 2 times per layer) it hits the 100% processor mark (checked using the top command on a SSH session) and the printing hangs for a couple seconds. 

I did find and experimental that's not quite a fix, but it may shed a light on the bug. If using the printer LCD display (reprap discount version) I change the feed rate to 90% it fixes. No more hangs EVEN if I then return it back to 100%. Tried to remove camera, add camera (it worked before the update to 1.10), change usb hub, change usb cable, direct connection to no avail. Only changing the feed rate after the printing started solved the issue.

GCODE and Systeminfo Bungle attached.

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

I did test it. Same problem.

### Version of OctoPrint

Octopi 0.18 - Octoprint Version 1.10.2

### Operating system running OctoPrint

Linux - Octopi 0.18

### Printer model & used firmware incl. version

Voron 1.5 - FIRMWARE_VERSION:v0.10.0-214-g98af5dff-dirty FIRMWARE_NAME:Klipper

### Browser and version of browser, operating system running browser

Chrome - Latest version

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[bags_gcode.zip](https://github.com/user-attachments/files/15991157/bags_gcode.zip)
[octoprint-systeminfo-20240626120258.zip](https://github.com/user-attachments/files/15991159/octoprint-systeminfo-20240626120258.zip)

cp2004, 2024-06-26:
Just wanted to check, since OctoPrint doesn't actually do anything with the feed rate value except send a command to the printer, that is is definitely the OctoPrint process using 100% and not the Klipper process using the CPU cycles?

If it's OctoPrint's side, I would expect it's something to do with the API or gcode sending loop here, not specifically tied to the feed rate.

andrebstv, 2024-06-26:
Hello @cp2004 I've checked using the 'top' command on a ssh terminal. The process name is indeed octoprint and not the klipper process. And it does indeed hit the 100%. And it visually coincides with the printer stuttering/hanging. 

I've checked the klippy log, and used the "Analyse" button. It indeed shows that the buffer is getting jammed to 100%. And another log, one I've used the "feedrate trick", shows a normal buffer, under 20%. Any idea how we can pinpoint this ?

Additional information: The hardware is a raspberry pi 3+ controlling a Arduino Mega + Ramps 1.4 board.

jneilliii, 2024-06-26:
One way to circumvent the issue might be to just set the feed rates in your slicer's start gcode maybe with `M220 S<percent>`

andrebstv, 2024-06-26:
@jneilliii that's what I did. Though it's quite clumsy jerry-rig.

andrebstv, 2024-06-27:
Now sometimes when the printing starts, before the M220 jerry rig, I get the timer too close error, and a message stating that the host might be running other processes... 

But it does not, pretty much exclusively runs octoprint...

foosel, 2024-06-27:
Just to quickly chime in, I've seen this, but I currently have no clue what's going on there and not even an idea how to debug this further. Will have to think about it a bit.

Certainly a new one.

foosel, 2024-06-27:
Just to add: the only thing I could imagine that would maybe give us more context right now is if you enabled the serial.log and reproduce the issue, then share a new system info bundle. I have the feeling there's something wonky happening on the serial line, and seeing what exactly is going on there could maybe shed some light.

andrebstv, 2024-07-08:
Ok now I`ve managed to reproduce the issue even using the M220 Jerry rig. It seems to happen more often when the part has circles.

Attached are the logs, including @foosel serial log, some other logs that had roughly the same time stamp as the serial log, and also the system bundle (generated a couple minutes after the print).

See attached HTOP print showing the processor hitting the high mark ( @cp2004 )
My device is a Pi3 with a nice heatsink case, and it doesn`t seem to be overheating.
[octoprint-systeminfo-20240707211703.zip](https://github.com/user-attachments/files/16121455/octoprint-systeminfo-20240707211703.zip)
[octoprint-logs.zip](https://github.com/user-attachments/files/16121457/octoprint-logs.zip)
[image]

andrebstv, 2024-07-29:
So no further checks on this issue ?

foosel, 2024-08-12:
If you have following the [blog](), there was some very bad crap to handle that took up all of my time, and after that I (thankfully!) had a desperately needed vacation scheduled. Some patience please, this project only has one maintainer (me).

foosel, 2024-08-14:
I've now spent several hours on going through the logs and also reading klipper source, and I still have no clue what is going on here.

The `serial.log` looks normal, and the klipper code doesn't look like changing the feedrate there does anything weird either.

Next step would be poking at things with a debugger.

I'd like you to try the following. First of all, install [py-spy](https://github.com/benfred/py-spy) on the server that OctoPrint is also installed on. I see you are running it on OctoPi, so this should do it:

``` bash
$ ~/oprint/bin/pip install py-spy
```

Confirm it is installed and usable by calling it:

``` bash
$ ~/oprint/bin/py-spy
py-spy 0.3.14
Sampling profiler for Python programs

USAGE:
    py-spy <SUBCOMMAND>

OPTIONS:
    -h, --help       Print help information
    -V, --version    Print version information

SUBCOMMANDS:
    record    Records stack trace information to a flamegraph, speedscope or raw file
    top       Displays a top like view of functions consuming CPU
    dump      Dumps stack traces for a target program to stdout
    help      Print this message or the help of the given subcommand(s)
```

Now, the next time you are observing the cpu spike, fetch the PID of OctoPrint:

``` bash
$ pid=$(ps -ef | grep -i octoprint | grep serve | tr -s " " | cut -d" " -f 2)
```

and use that to run `py-spy top`:

``` bash
$ ~/oprint/bin/py-spy top -p $pid -r 50 --full-filenames
```

**Warning**: This might negatively impact the print, so only do this with something you can live with to have to print again. 

This should print out something like this:

```
Collecting samples from '/home/pi/oprint/bin/python3 /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000'
Total Samples 300
GIL: 0.00%, Active: 2600.00%, Threads: 46, Processes 1

  %Own   %Total  OwnTime  TotalTime  Function (filename)                                                                                                                                                                                                                                                                      
2300.00% 2300.00%   138.0s    138.0s   _worker (/usr/lib/python3.7/concurrent/futures/thread.py)
100.00% 100.00%    6.00s     6.00s   _ssdp_monitor (/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery/__init__.py)
100.00% 100.00%    6.00s     6.00s   _thread (/home/pi/oprint/lib/python3.7/site-packages/sentry_sdk/sessions.py)
100.00% 100.00%    6.00s     6.00s   read_events (/home/pi/oprint/lib/python3.7/site-packages/watchdog/observers/inotify_c.py)
  0.00%   0.00%   0.040s    0.100s   _get_value (/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings/__init__.py)
  0.00%   0.00%   0.020s    0.020s   _release_save (/usr/lib/python3.7/threading.py)
  0.00%   0.00%   0.020s    0.040s   <listcomp> (/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings/__init__.py)
  0.00%   0.00%   0.020s    0.020s   __iter__ (/home/pi/oprint/lib/python3.7/collections/__init__.py)
  0.00%   0.00%   0.020s    0.140s   refresh_serial_list (/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/__init__.py)
  0.00%   0.00%   0.020s    0.120s   serialList (/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py)
  0.00%   0.00%   0.020s    0.060s   wrapper (/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py)
  0.00%   0.00%   0.000s    0.100s   get (/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings/__init__.py)
  0.00%   0.00%   0.000s    0.160s   run (/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py)
  0.00%   0.00%   0.000s    0.040s   _cached_prefixed_keys (/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings/__init__.py)
  0.00%   0.00%   0.000s    0.040s   get_by_path (/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings/__init__.py)
  0.00% 2500.00%   0.000s    150.0s   run (/usr/lib/python3.7/threading.py)
  0.00%   0.00%   0.000s    0.020s   __iter__ (/home/pi/oprint/lib/python3.7/_collections_abc.py)
  0.00% 100.00%   0.000s     6.00s   run (/home/pi/oprint/lib/python3.7/site-packages/watchdog/observers/inotify_buffer.py)
  0.00%   0.00%   0.000s    0.040s   _has_prefix (/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings/__init__.py)
  0.00%   0.00%   0.000s    0.020s   wait (/usr/lib/python3.7/threading.py)
  0.00% 2600.00%   0.000s    156.2s   _bootstrap (/usr/lib/python3.7/threading.py)
  0.00%   0.00%   0.000s    0.100s   process (/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings/__init__.py)
  0.00% 2600.00%   0.000s    156.2s   _bootstrap_inner (/usr/lib/python3.7/threading.py)
  0.00% 2400.00%   0.000s    144.2s   run (/home/pi/oprint/lib/python3.7/site-packages/sentry_sdk/integrations/threading.py)
```

and hopefully allow us to get a better idea of *where* it's spending all of these CPU cycles.

foosel, 2024-11-18:
@andrebstv Have you tried the above? I'm waiting on your findings here.

closed 2025-02-05 by foosel