OctoPrint/OctoPrint #5035: Could not initialize event manager: non-string found in code slot

EveryOtherUsernameWasAlreadyTaken, 2024-06-28:
### The problem

After upgrading to 1.10.2, Octoprint does not start any more. The website says I should log in and check the logs.

Logging in via SSH, I get the following message in `~/.octoprint/logs/octoprint.log`:

```none
2024-06-28 14:24:22,640 - octoprint.startup - INFO - ******************************************************************************
2024-06-28 14:24:22,644 - octoprint.startup - INFO - Starting OctoPrint 1.10.2
2024-06-28 14:24:22,652 - octoprint.startup - INFO - ******************************************************************************
2024-06-28 14:24:22,786 - octoprint.startup - CRITICAL - Could not initialize event manager: non-string found in code slot
2024-06-28 14:24:22,787 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
```

Rebooting does not help.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

sudo: octoprint: command not found

### Version of OctoPrint

1.10.2 (there is no corner to look that up if the webserver does not start)

### Operating system running OctoPrint

Linux octopi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux, version of /etc/octopi_version: 1.0.0

### Printer model & used firmware incl. version

Prusa i3 MK3

### Browser and version of browser, operating system running browser

Firefox 127.0.2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

## Systeminfo Bundle

I can't include the Systeminfo bundle, if it's only accessible via the web interface. The web interface does not work.

The recovery page is exactly the same as the normal page. I can't download Systeminfo bundle there either.

I can't generate systeminfo bundle on command line, because octoprint cannot be found

```bash
pi@octopi:~ $ octoprint safemode
-bash: octoprint: command not found
pi@octopi:~ $ sudo !!
sudo octoprint safemode
[sudo] password for pi:
sudo: octoprint: command not found
```

## Javascript Console

```
This page is in Quirks Mode. Page layout may be impacted. For Standards Mode use “<!DOCTYPE html>”.
```

Note: this is the page that says that the webserver cannot be started. That page does not do much.

## Screenshots

[image]

[image]

[image]

## GCode

It's not a GCode problem. I can't even start printing.

EveryOtherUsernameWasAlreadyTaken, 2024-06-28:
Dear Bot, I have explained in detail why it isn't possible for me to upload the requested information. I have tried all three methods explained on the page that deals with that information. Please do not close the issue for this reason. 

If that information is so super important, then please make sure that it can be accessed also in ever so severe cases.

MarshMaker3D, 2024-06-28:
Just a shot in the dark, but maybe rule out corrupt image and/or SD card?

EveryOtherUsernameWasAlreadyTaken, 2024-06-28:
I formatted the SD card and reinstalled everything from scratch. Raspberrys and/or SD cards suck. Yet another hour wasted.

closed 2024-06-28 by EveryOtherUsernameWasAlreadyTaken

jneilliii, 2024-06-28:
Recommend the backup scheduler plugin and one of the cloud based backup plugins to expedite setting up from scratch.