OctoPrint/OctoPrint #5036: Wonky Trusted Downstream Validation

Silvenga, 2024-06-28:
### The problem

When using a reverse proxy within another network, Octoprint appears to incorrectly validate the `X-Forwarded-For` header.

---

In my setup, I have the reverse proxy in `172.24.0.0/24`. The reverse proxy is proxing for the `192.168.0.0/24` network.

After configuring a static IP on my reverse proxy and configuring that IP in the trusted proxies list of Octoprint (allowing a subnet work be nice, but this is unrelated), Octoprint incorrectly handles the `X-Forwarded-For` list.

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

Offical container image, running on Ubuntu 22.04

### Browser and version of browser, operating system running browser

Firefox 127.0.2 on Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240628132433.zip](https://github.com/user-attachments/files/16034570/octoprint-systeminfo-20240628132433.zip)

jneilliii, 2024-06-28:
> Offical container image, running on Ubuntu 22.04

so 172.24.0.0/24 is your docker network range?

Silvenga, 2024-06-28:
Yep, all host-local containers are in that network.

Silvenga, 2024-06-28:
```python
import netaddr

def get_http_client_ip(remote_addr, forwarded_for, trusted_proxies):
    try:
        trusted_ip_set = netaddr.IPSet(trusted_proxies)
    except netaddr.AddrFormatError:
        # something's wrong with one of these addresses, let's add them one by one
        trusted_ip_set = netaddr.IPSet()
        for trusted_proxy in trusted_proxies:
            try:
                trusted_ip_set.add(trusted_proxy)
            except netaddr.AddrFormatError:
                print(
                    f"Trusted proxy {trusted_proxy} is not a correctly formatted IP address or subnet"
                )

    if forwarded_for is not None and sanitize_address(remote_addr) in trusted_ip_set:
        for addr in (
            sanitize_address(addr.strip())
            for addr in reversed(forwarded_for.split(","))
        ):
            if addr not in trusted_ip_set:
                return addr
    return sanitize_address(remote_addr)

def sanitize_address(address):
    address = unmap_v4_as_v6(address)
    address = strip_interface_tag(address)
    return address

def strip_interface_tag(address):
    if "%" in address:
        # interface comment, e.g. "fe80::457f:bbee:d579:1063%wlan0"
        address = address[: address.find("%")]
    return address

def unmap_v4_as_v6(address):
    if address.lower().startswith("::ffff:") and "." in address:
        # ipv6 mapped ipv4 address, unmap
        address = address[len("::ffff:") :]
    return address

client_ip = get_http_client_ip("172.24.0.2", "192.168.0.148,172.24.0.2", ["172.24.0.2"])

print(client_ip) # 192.168.0.148
```

Looking at the code, everything looks fine... I'm not sure where this discrepancy is coming from.

Silvenga, 2024-06-28:
Oh, I see it, `trustedUpstream` vs `trustedDownstream`, I think there's a typo.

Silvenga, 2024-06-28:
Okay, figured it out, related to `get_http_client_ip`:

https://github.com/OctoPrint/OctoPrint/blob/ad9ebcaa0ef91950652fb2561b7c06b79d95a455/src/octoprint/util/net.py#L302-L322

With the Octoprint docker container, there always exists HAProxy (which wasn't obvious without looking at the container image) - so the client IP is always `127.0.0.1`, this will cause `sanitize_address(remote_addr) in trusted_ip_set` to always noop if `trustedDownstream` is ever configured.

https://github.com/OctoPrint/OctoPrint/blob/ad9ebcaa0ef91950652fb2561b7c06b79d95a455/src/octoprint/util/net.py#L316-L322

The complete workaround is to configure trusted proxies like so:

```yaml
server:
  reverseProxy:
    trustedDownstream:
      - 172.24.0.2
      - 127.0.0.1
    trustedUpstream:
      - 172.24.0.2      
      - 127.0.0.1
```

That said, I think it would be best to always trust the loopback network `127.0.0.0/8`, as well as unify `trustedUpstream`/`trustedDownstream` (assuming this was a typo).

```python
trusted_ip_set.add("127.0.0.0/8")
```

foosel, 2024-08-12:
My July got completely eaten up by the stats manipulation, and after that I really needed a vacation (which thankfully had already been scheduled). Hence I'm only now getting a chance to look into this, and I've now added it to my TODO list for the week. Thank you for the amazing amount of analysis you put into it!

Silvenga, 2024-08-12:
> @foosel My July got completely eaten up by the stats manipulation, and after that I really needed a vacation (which thankfully had already been scheduled). Hence I'm only now getting a chance to look into this, and I've now added it to my TODO list for the week. Thank you for the amazing amount of analysis you put into it!

Yeah, I figured! No worries!

To be clear, I don't believe this to be any type of security issue, after understanding that the Octoprint container image internally uses an reverse-proxy.

I think the ultimate fix for this (not including the config typo) might be to improve the defaults (maybe a different default in the container image, or including RFC 1918 private ranges by default) or document the wonky'ness (e.g. the reverse proxy existence, noting that subnets are supported, etc.). 

IMO, as someone who deals with this stuff a lot as both a developer and a DevOps engineer, Octoprint's implementation of trusted proxies is a little confusing (although, it does protect against bad configurations, so the original goals are met). Most of the confusion is around needing to specify `127.0.0.1/8` as a trusted proxy.

foosel, 2024-08-15:
> I think the ultimate fix for this (not including the config typo) might be to improve the defaults (maybe a different default in the container image, or including RFC 1918 private ranges by default) [...]

I agree that there's been some wonkyness, and I hopefully have that now fixed.

`trustedDownstream` will be migrated to `trustedProxies` (and `trustedUpstream` removed - the typo has been fixed) and also `trustLocalhostProxies`. If the latter is true (the default), whatever is defined in `trustedProxies` will also get `127.0.0.0/8` and `::1` prepended for evaluation.

The docs have been adjusted accordingly, as has the FAQ entry.

I think the source of confusion here primarily was that OctoPrint would always use the list as-is and default to `127.0.0.1` and `::1`. So if you didn't touch the settings, all would work with a reverse proxy on the same machine, but if you wanted to add additional proxies in front, you'd have to explicitly add the localhost addresses as well. That's how all of the lists in OctoPrint's settings work, they are never cumulative, as that would prevent someone from having full control over the content vs the default, but here that caused some head scratching.

I hope that things are now clearer by splitting things up between "here's a list of (additional) proxies" and "trust localhost".

foosel, 2024-08-15:
I've also just back ported the bug fix (`trustedUpstream` -> `trustedDownstream`) to the bugfix branch, so should another 1.10.x bugfix release go out, that will included there.

The rest will be shipped with 1.11.0.

commit bb12d7d7 (2024-08-15): config: change trustedDownstream to trustedProxies + flag

As outlined in #5036, trusted proxy configuration was a bit confusing up
until now. localhost wasn't added automatically in case additional proxies
were configured, leading to hard to debug issues in the field.

Additionally, there was some confusion with regards to trustedDownstream
vs trustedUpstream, the latter being a typo that crept into the code.

The setting has now been renamed to trustedProxies. An additional flag
trustLocalhostProxies defaulting to True will ensure that localhost 127.0.0.0/8
and ::1 will always be trusted, if [... cut]
closed 2025-04-22 by foosel