OctoPrint/OctoPrint #5037: Emergency stop button possibly as a prominent but by-default-disabled feature?

jaggzh, 2024-07-03:
### Is your feature request related to a problem? Please describe.

Emergency Stop (like in Simplify3d) has saved me many times when the printer has done something wrong. Just yesterday my printer's Z-home endstop failed to trigger and my head kept driving down into the bed.

### Describe the solution you'd like

Re: https://github.com/OctoPrint/OctoPrint/issues/2071
Due to:
1. The concerns of it providing a "false sense of security"
2. Custom controls being an inappropriate place for an "Emergency Stop" button
3. The valuable nature of it for many users

I'm thinking a good balance might be to either:
1. Implement the prominent serious Emergency Stop button available globally, but have it default disabled.
2. Expand custom controls to allowing globally-available controls.

### Describe alternatives you've considered

There is really no alternative. I believe that's why there's that big button on tons of machinery/shop equipment -- even if it doesn't solve every problem that might come up.  It has saved people's lives and the lives of many innocent or vindictive machines who later only needed therapy. :}  In my case, even for my printer, it's been used many times, and served its purpose very well.

Oh, and thank you for all your amazing work though!  <3  Octoprint is a beautiful project.

cp2004, 2024-07-03:
Have you considered/tried out the emergency stop plugins? By utilising the plugin system, users who *want* a prominent emergency stop can have one, those that don't have no idea.

jaggzh, 2024-07-05:
Okay. I got that working. It had some error accessing an asset when I first
added it. But restarting octoprint and it's there. (I've not yet tested it.
Need to repair and level my printer and whatnot from the z driving with
into the bed until I shut the printer off). I don't actually understand the
reasons why it's considered a bad feature to always have available for
everyone, but hopefully the plug-in works for little ol' me. Thanks a
bunch!

closed 2024-07-06 by jaggzh