OctoPrint/OctoPrint #5038: [1.11.0.dev] Exception when saving changed plugin repo URL

foosel, 2024-07-04:
### The problem

```
2024-07-04 11:41:06,742 - octoprint.server.util.tornado - ERROR - Exception in WSGI application
Traceback (most recent call last):
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/server/util/tornado.py", line 766, in handle_request
    app_response = await loop.run_in_executor(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/flask/app.py", line 2531, in wsgi_app
    response = self.handle_exception(e)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/flask/app.py", line 2528, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/server/util/flask.py", line 1645, in decorated_view
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/vendor/flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
         ^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/server/api/settings.py", line 472, in setSettings
    response = _saveSettings(data)
               ^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/server/api/settings.py", line 1259, in _saveSettings
    s.save(trigger_event=True)
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/settings/__init__.py", line 1778, in save
    "effective_hash": self.effective_hash,
                      ^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/settings/__init__.py", line 1002, in effective_hash
    hash.update(self.effective_yaml.encode("utf-8"))
                ^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/settings/__init__.py", line 992, in effective_yaml
    return yaml.dump(self.effective)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/util/yaml.py", line 95, in dump
    return _save_to_file_base(data, file=None, pretty=pretty, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/devel/OctoPrint/OctoPrint/src/octoprint/util/yaml.py", line 59, in _save_to_file_base
    return yaml.dump(
           ^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/__init__.py", line 253, in dump
    return dump_all([data], stream, Dumper=Dumper, **kwds)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/__init__.py", line 241, in dump_all
    dumper.represent(data)
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 27, in represent
    node = self.represent_data(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 48, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 207, in represent_dict
    return self.represent_mapping('tag:yaml.org,2002:map', data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 118, in represent_mapping
    node_value = self.represent_data(item_value)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 48, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 207, in represent_dict
    return self.represent_mapping('tag:yaml.org,2002:map', data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 118, in represent_mapping
    node_value = self.represent_data(item_value)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 58, in represent_data
    node = self.yaml_representers[None](self, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gina/.pyenv/versions/3.11.2/envs/octoprint-py311/lib/python3.11/site-packages/yaml/representer.py", line 231, in represent_undefined
    raise RepresenterError("cannot represent an object", data)
yaml.representer.RepresenterError: ('cannot represent an object', <StylesheetEnum.css: 'css'>)
```

### The solution

Needs to be investigated further, likely caused by the recent switch to pydantic 2 in some way or another.

@cp2004 does this maybe look like the issues you were seeing? Can't understand right now why I was not being able to reproduce something like that earlier.

cp2004, 2024-07-04:
> @cp2004 does this maybe look like the issues you were seeing? Can't understand right now why I was not being able to reproduce something like that earlier

Yes

foosel, 2024-08-14:
Fixed by the above commit.

commit 2f862f95 (2024-08-15): fix: saving settings

Pydantic's base models also needed to be configured with
validate_default=True for enums to correctly convert to str
even when used as default values.

Closes #5038
closed 2024-08-14 by foosel