OctoPrint/OctoPrint #5039: [Request] Improve preview of files using "scarf joint seam" (orcaslicer & others)

jepler, 2024-07-05:
### Is your feature request related to a problem? Please describe.

I've recently started to use the "scarf joint seam" feature of orcaslicer, which can improve the surface finish of many 3d prints.

However, this feature emits XYZE moves which confuse the gcode preview in octoprint, making it report an incorrect number of layers, as well as making the layer-by-layer preview not useful.

### Describe the solution you'd like

Octoprint needs to use a different method to analyze gcode files that use scarf join. However, I don't know enough about the internals to have a concrete suggestion.

### Describe alternatives you've considered

Many slicers emit special comments to delimit the slicer's idea of a layer. Octoprint could (optionally?) look for these in lieu of what it appears to do now, analyze the Z-heights where extrusion takes place. 

### Additional context

I can provide a gcode file reproducer if needed. I'm using OrcaSlicer 2.1.1 on Linux.

This might also improve gcode visualization of vase mode prints, as well.

jneilliii, 2024-07-05:
This is an interesting challenge. I know @JoveToo was one of the last contributors to the GCode Viewer, wonder if he might have some ideas.

JoveToo, 2024-07-05:
Please upload a gcode file from a benchie or something with scarf joint seam.

I will take a look.

jepler, 2024-07-05:
I've updated the initial comment with a zipped gcode file that demonstrated the problem.
> Model size: 154.79mm × 123.30mm × 48.00mm
> Estimated layer height: 0.00mm
> Estimated total print time: 00:49:52
> Layers with extrusion: 13327