OctoPrint/OctoPrint #5042: [Request] Support arbitrary url for new camera stack to support containers

poehlert, 2024-07-12:
### Is your feature request related to a problem? Please describe.

For a variety of reasons, I run stock Raspberry Pi OS with docker containers for octoprint and the camera stack. I have a container for octoprint, a container for the camera process, and some other containers that are configured using docker compose. I've switched my camera container to the new camera streamer stack, but am disappointed that I can't enable webrtc in octoprint for lower bandwidth/latency because the camera-streamer url appears to be hard coded to http://127.0.0.1/webcam. 

### Describe the solution you'd like

In the camera settings support entering the address to camera streamer when enabled: [image]

### Describe alternatives you've considered

I currently am using the legacy camera stack and MJPEG which is really chunky.

### Additional context

I also tried to look at the Camera-Streamer-Control plugin (https://plugins.octoprint.org/plugins/camerastreamer_control/) hoping that maybe it would expose such a configuration but it does not.

jneilliii, 2024-07-12:
Looks lower in the classic webcam plugin settings. That's where you can configure the URLs.

cp2004, 2024-07-12:
@jneilliii is correct, OctoPrint now supports different webcam streaming implementations using plugins. If you install the camera streamer control plugin, you can find its configuration along with all the other plugins in your settings. That should work fine for you.

poehlert, 2024-07-12:
OK, thank you. In writing up a response with detailed snapshots, I'm seeing now in the camera streamer control plugin where I can specify the url for my camera-streamer container. Thanks for the help, and I have it working now.

closed 2024-07-12 by poehlert