OctoPrint/OctoPrint #5043: Ootoprint  1.10.2 and  Custom Control Editor

Thisismydigitalself, 2024-07-14:
### The problem

Just upgrded OP 1.9 to 1.10.2 and Custom Control Editor  buttons are missing, i had this issue a year ago when upgraded OP version  but then uninstalling and reinstalling the plug fixed the issue.

Not this time.
No matter what i try - custom control editor plug does not work - it loads but buttons are missing.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Delta 

### Browser and version of browser, operating system running browser

Win10/Firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Does SystemInfo files include any private information?

jneilliii, 2024-07-14:
This sounds like the situation where that plugin wipes out your custom control settings occasionally due to a race condition. It's recommended to disable that plugin once you setup the controls the way you like it, they will still function, to prevent the plugin from clearing those settings out.

jneilliii, 2024-07-14:
And since this is technically a plugin issue and not OctoPrint issue, you should report the issue to the plugin's repo, not here.

closed 2024-07-14 by jneilliii

cp2004, 2024-07-15:
If you still see the buttons in your config.yaml, then you should try safe mode to see if they come back. They're built in to OctoPrint and so you don't need the plugin to view them.

If you're having issues, please use the community support channels such as https://community.octoprint.org and https://discord.octoprint.org. If the consensus then points to a bug in OctoPrint, then we can come back to fix it here.

Thisismydigitalself, 2024-07-16:
> you should try safe mode 
tried that to no avail.

jneilliii, 2024-07-16:
probably need a look at the browser's developer tools for any errors. as cp2004 mentioned, we can probably give near real-time assistance at https://discord.octoprint.org/

Thisismydigitalself, 2024-07-17:
UPDATE:
i renamed my config.backup with has my buttons to config. yaml, than restarted octoprint and just before reloading the UI i saw the buttons grayed out but once UI reloaded - they are not there anymore. i will try to screenshot my screen and upload. i really hope this is something related to octoprint which you guys can fix this.

Thisismydigitalself, 2024-07-18:
UPDATE:
i renamed my config.backup which has my buttons to config. yaml, than restarted octoprint and just before reloading the UI i saw the buttons grayed out but once UI reloaded - they are not there anymore. i will try to screenshot this and upload. i really hope this is something related to octoprint which you guys can fix.
once the server reloaded - the config.yaml file removes the commands/buttons related to custom control editor plugin, AND the new updated config.backup DOES INCLUDES these. for me it seems like octoprint corrupts my yaml file.