OctoPrint/OctoPrint #5044: GCode Viewer hangs with Firefox 128.0 64bit

getgray, 2024-07-14:
### The problem

repeat of #4879 

The problem:

Large gcode files (> ~23 MB test file) loaded into OctoPrint will "parse" to a given number (depending on the gcode; it seems to be consistent per file, but varies between files), 81% in this test case, and then stop. Smaller gcode files untested.

Test file worked in Chrome. No content blocking installed.

This was failing with a previous version of FF, but wsa fixed as reported in issue #4879.  Has been workign until I send this large file today, it's back to failing with FF.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

Octopi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

getgray, 2024-07-15:
By starting the gcode viewer immediately when the print starts, the viewer loads, albeit very slowly.  That workaround is good enough, closing bug request.

foosel, 2024-07-17:
In any case, this sounds like it is rather a browser bug, so not something I can really fix. If fetching a resource doesn't work, it doesn't work, there's nothing I can do to make it work. I wonder what broke though... It took long enough to see the original issue get fixed on the released browser versions...