OctoPrint/OctoPrint #5046: [Request] Enable a couple of things of the gcode viewer even when print is too big.

rewolff, 2024-07-18:
### Is your feature request related to a problem? Please describe.

I am currently wondering: Is my print on the very last layer? How long does a layer of this print take? 

Those are things that the "gcode viewer" tab would normally show, but this time the print file is "too big" and indeed, my browser slows down if I "power through", so currently not in the mood to click please visualize ... regardless of its size. 

### Describe the solution you'd like

The actual visualization should "not work". But processing the file, calculating total print time, and layer times could proceed as normal. Now the normal visualization window could show the message: "File too big to visualize, click here to load anyway" while "model info" and "layer info" panels work as normal. (and "render options" panel could be grayed out, or be used for the file too big message.

rewolff, 2024-07-18:
Just FYI: the "normal processing" is not entirely normal: The real normal processing would store the parsed gcode somewhere in memory. The "big file mode" would parse-and-calculate things but NOT store. So it would need to process a large "stream" of data, but not use a lot of memory. (On a big print of 400mm there are only 2000 layers. Peanuts. Maybe 4000 with extremely small layer height. Still peanuts.

cp2004, 2024-07-18:
It's not the actual visualisation that is intensive on the browser, but the loading & parsing of the gcode file. I am not sure that it would be possible to work out layer durations and other stats without processing the file, at least not without major work on the architecture of the gcode viewer.

You can get basic stats such as total time & model size from the file list, from the processing done server side. Plugins such as print time genius make this more accurate, and Display Layer Progress can show you the current layer as well. As they're server side implementations they don't suffer in the browser the same way the gcode viewer does.

cp2004, 2024-07-18:
I don't know how closely you've dug into the gcode viewer, but normally once the file is 'analysed' then there is not a large resource use. There was a lot of work done to introduce compression for the data structure, which you can turn on - it means the processing takes longer, but the memory footprint afterwards is lower. My browser only "slows down" whilst this processing is taking place, indicated in the UI, then once this is done it seems fine.

Is this your experience?

cp2004, 2024-07-18:
To add, we did look into doing the analysis on-the-fly as it was required, but this was too slow so that scrolling through the layers etc. was very difficult to use.

rewolff, 2024-07-21:
That last one is a whole different level of "on the fly" . What I was thinking of was just analysing each layer to the level that you have a <X fillament, Y time on this layer > so that the "this layer" panel can be filled. 

I am of the opinion that "more memory" is more likely to speed up user-experience than a faster CPU. So when buying a new PC I always buy lots of ram and a high mid-range CPU. (before you start paying premium for "one of the fastest CPUs"). 

So in the past I was "I should have plenty of RAM, so I'm not bothered by the warning. My GCODE file is 26Mb, I have 1000 times more RAM (on one machine and 2000x on the other) so go ahead and load it.  

But then I started noticing that my browser sometimes slows down so that videos become "quick slideshows" with only about 5 fps or so.... restarting the browser works for a while.  It doesn't happen immediately. But since I've stopped thinking: "I should have plenty of RAM, goahead and load it", this slideshow effect seems to have occurred less often. 

It'd be nice and easy-to-prove when restarting the browser fixes the videos, and then loading the big file immediately causes the videos to slow down again. Alas, it is not that direct. Maybe it is something entirely different and I'm blaming the wrong culprit. 

But what you are saying is: don't worry you should be fine loading big files. 
Then the big warning should be removed.

cp2004, 2024-07-23:
> But what you are saying is: don't worry you should be fine loading big files.

That's not what I was going for - apologies if I didn't explain it clearly.

My point is, have you checked that it is actually using high memory during the visualisation and this is causing your issue?

When you open the gcode viewer, there is a progress bar along the bottom while it analyses the file. Do you only experience slowdown while this is processing, or does it continue the whole time?

Does the same occur with & without compression enabled?

Lastly, because it is a significant rework of the architecture of the gcode viewer it's probably unlikely someone will implement this, they may rather focus on improving the performance of the analysis itself.

rewolff, 2024-07-23:
Do you think I enjoy making my browser unusably slow? 

There is this big full-window warning. You are implying that warning is outdated. In that case, remove the stupid warning. 

Or the warning might be there because the described problem actually happens. I don';t care if it actually happens to me. I've "ignored the big warning in the past" and "my browser has slowed down in the past". I've switched to not clicking past the "big warning" and "my browser doesn't slow down as often as it used to". For now I;m sticking to not clicking "please visualize regardless of its size". 

That these slowdowns were related to "load anyway" has taken me a while to figure out: The effect is not immediate. 

In that case I have provided a suggestion to make the software more usable. 

If the info panel does an "analyse this layer" on the stored data to get the layer info once I "scroll" to a layer, then adding an "if cached data exists return cachec data" somewhere should be easy and not have a large impact on the code. 

And then there are small incremental steps to just analyse, cache the layer info and then discard the data.

cp2004, 2024-07-23:
My questions were aimed at you _testing_ this, to try and determine whether large files in the gcode viewer are actually causing your issue. If your browser still slows down at all _without_ visualising the gcode, & you can't notice a definite issue with loading (which is the heaviest part of the process) then it's likely not that.

I don't want to chase solutions that don't have a problem, which is why I was trying to narrow it down. The larger the file, the more processing it takes. The warning stays because there is a limit somewhere, and while we've worked to increase performance and the limit has generally risen, different people have different spec machines. That's why it's your choice, not ours to run the gcode viewer. You can configure the file size threshold if you know you can increase it.

rewolff, 2024-07-23:
I /have/ increased it already.

rewolff, 2024-07-23:
OK. I tested the following: 
loading the gcode increases one of the chrome render processes by 400Mb  (from 250 to 650 MB resident RAM). 
Enabeling force compression makes the increase only 20Mb. In both cases the loading/analysing takes about 10 seconds. 
With the 400Mb "original" I tried if I got the slowdown and I did not get that immediately. That's why it was difficult to associate with the loading of big files. With a 64Mb machine, the 400Mb seems quite wasteful: The gcode itself is only 26.7 Mb. Apparently analyzing it, throwing away unnecessary info and storing the useful info leads to a 15 fold increase in the required amount of storage. Oh well.

rewolff, 2024-08-04:
OK. Today I have a big print going. I was monitoring it on one screen and watching videos on the other. To reduce my bandwidth usage I switched from the camera view to the gcode viewer, hit "force compress" and "load anyway". 

I am now typing this faster than my browser can keep up putting the letters in the box. And videos are "slideshow" again. So yes, the slowing down is real and still happens even when you click "force compression".