OctoPrint/OctoPrint #5048: Filament run out does not allow resuming anymore

mattbodley, 2024-07-26:
### The problem

After the upgrade to the new version, I noticed that the resume button is no longer in the UI anymore after the filament sensor detects a filament run out and I change it. It just pauses and does nothing.

[octoprint-systeminfo-202
[octoprint.log](https://github.com/user-attachments/files/16386053/octoprint.log)
40726035924.zip](https://github.com/user-attachments/files/16386046/octoprint-systeminfo-20240726035924.zip)
[image]

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

I can test in safe mode and will do so when the next roll gets closer to the end, but I just don't think that the issue will be solved this way as filament run out is a core feature and not a plugin feature

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

OctoPi 1.0.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

mattbodley, 2024-08-03:
Ok, just ran Octoprint in safe mode and the filament run out sensor resumes fine!!! 

But does that mean that the core is working? Because the core system shouldn't allow a plugin to do that, right? So I'm thinking that a plugin is the result of it not working, but it is due to the core allowing a plugin to do something it is not supposed to do?

Any ideas on how to find out "which" plugin is causing the issue? Apart from the slow process of disabling them all and turning them on one by one doing the whole process of filament changing

jneilliii, 2024-08-03:
This definitely means plugin issue. To figure out which one disable half of your plugins and test. If it still works then it was one of the disabled ones, if not it's one of the remaining enabled. Disable or enable another half based on results, rinse and repeat.

foosel, 2024-08-15:
Closing as this is caused by a third party plugin.

closed 2024-08-15 by foosel