OctoPrint/OctoPrint #5052: MK4 with Octoprint - M486 A<long name> crashes the printer

DerMouse, 2024-08-29:
### The problem

PruscaSlicer writes « M486 A<stuff> » into the gcode file. If line is longer than 83 characters, sending the file to print through Octoprint USB connection will first reboot the printer then crash it. Printer must then be restarted. No problem if same file printed through regular USB key.

**Note** : I did report this initially on prusa3d/Prusa-Firmware-Buddy (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4123) thinking it was a firmware bug, but it seems it comes from Octoprint (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4123#issuecomment-2318481029).

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

Linux Gentoo

### Printer model & used firmware incl. version

Prusa MK4 fw 6.0.1+14848

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Offending GCODE
[M486.txt](https://github.com/user-attachments/files/16803975/M486.txt)

Serial log file when crashing
[serial.log.txt](https://github.com/user-attachments/files/16803976/serial.log.txt)

MK4 display after it reboots.
[image]

System bundle
[octoprint-systeminfo-20240830001314.zip](https://github.com/user-attachments/files/16806528/octoprint-systeminfo-20240830001314.zip)

foosel, 2024-08-30:
As explained in prusa3d/Prusa-Firmware-Buddy#4123, this *is* in fact an issue in the firmware, OctoPrint is doing everything correctly here. I'll continue to comment on the firmware ticket.

closed 2024-08-30 by foosel

bkerler, 2024-08-30:
@foosel is there a specific reason that the resend limit for already failed but numbered and checksumed messages defined as "maxConsecutiveResends" in octoprint is set to a default value of 10 ? Wouldn't a value much lower would make more sense, like 3 ?

bkerler, 2024-08-30:
Also if a communication is confirmed to fail .. does it make sense to send an emergency M112 instead of pausing the print and sending a user notification about it ?

jneilliii, 2024-08-30:
> does it make sense to send an emergency M112 instead of pausing the print and sending a user notification about it ?

If it's reported as an error sure. It only does that though if the option is enabled.

bkerler, 2024-08-31:
> 
> If it's reported as an error sure. It only does that though if the option is enabled.
> 
> 

Thanks. Then I assume it is expected behaviour and not an issue.