OctoPrint/OctoPrint #5056: Title attributes (in permission list in access control) are not translated

WildRikku, 2024-09-10:
### The problem

While the descriptions for permissions in the "edit user" view in the access control settings are translated, their title mouseovers are not. Investigating the code, it's clear why: The untranslated description is used as the title. I suspect this was either overlooked or ignored for a reason when working on #4626 which included https://github.com/OctoPrint/OctoPrint/commit/eb89369c6f96f855fc468d686cc54ffa7b251190 which introduced a Knockout binding to use translated strings in Jinja templates.

Shamelessly copying that commit, I tried to fix the issue by adding another Knockout binding to be able to use translated texts in attributes. It works, but it also somehow kills everything else. I attached the diff of my attempt (it's small and it does cause more trouble than it solves so I did not open a PR). Feel free to use it or help me learn more to do fix it myself. :) I tried to keep it generic so it's useful for the future.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.10.2+ga8fff3930.dirty

### Operating system running OctoPrint

SteamOS 3.5.19

### Browser and version of browser, operating system running browser

Firefox 130.0 Flatpak

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

```Less has finished and no sheets were loaded. [packed_libs.js:4080:1121](http://localhost:5000/static/webassets/packed_libs.js?409a9b49)
Starting dependency resolution... [packed_core.js:1858:521](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Calling OctoPrintClient.getRequestHeaders with additional headers as the first parameter is deprecated. Please consult the docs about the current signature and adjust your code accordingly. [packed_client.js:8:467](http://localhost:5000/static/webassets/packed_client.js?673073c5)
... dependency resolution done [packed_core.js:1871:5](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Initial application setup done, connecting to server... [packed_core.js:1894:867](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Connected to the server [packed_core.js:1684:456](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Triggering connect on all view models [packed_core.js:1689:240](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Server (re)connect processed [packed_core.js:1694:5](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Finalizing application startup [packed_core.js:1894:243](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Going to bind 44 view models... [packed_core.js:1875:39](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model FirstRunWizardViewModel to target #wizard_firstrun_start since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model FirstRunWizardViewModel to target #wizard_firstrun_end since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model ClassicWebcamWizardViewModel to target #wizard_classicwebcam since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model FileCheckViewModel to target #wizard_plugin_file_check since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_update since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_settings since it does not exist [packed_core.js:1887:44](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
User christian logged in [packed_core.js:262:907](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Login mechanism: remember_me [packed_core.js:267:35](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
... binding done [packed_core.js:1892:173](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Application startup complete [packed_core.js:1892:465](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
downloadable font: Glyph bbox was incorrect (glyph ids 1 3 4 5 6 7 9 12 22 24 25 29 30 33 34 35 36 37 38 39 45 46 48 49 50 51 53 54 55 58 60 64 65 66 67 71 72 73 81 86 87 88 91 93 100 101 105 106 109 112 113 130 133 140 141 142 146 148 149 151 153 154 155 162 165 170 175 180 181 188 191 193 194 195 196 198 199 201 204 205 212 213 215 216 221 223 225 227 228 232 243 245 246 248 250 251 254 255 257 258 261 262 266 269 270 273 275 276 277 280 281 284 287 291 292 295 300 303 306 307 308 313 314 316 319 321 324 325 327 328 330 334 335 336 337 342 345 348 349 351 357 359 360 362 366 369 371 372 373 374 376 377 381 384 387 388 389 396 398 399 400 401 416 417 420 425 426 428 432 433 437 439 440 441 444 449 454 456 457 458 461 464 469 470 471 472 475 477 480 481 483 485 486 487 488 490 491 496) (font-family: "Font Awesome 6 Brands" style:normal weight:400 stretch:100 src index:0) source: http://localhost:5000/static/vendor/fontawesome-6.1.1/webfonts/fa-brands-400.woff2
Connected to the server [packed_core.js:1684:456](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Login mechanism: remember_me [packed_core.js:267:35](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Triggering connect on all view models [packed_core.js:1689:240](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Server (re)connect processed [packed_core.js:1694:5](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Cookie-Warnungen 2
Login mechanism: password [packed_core.js:267:35](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
OctoPrint.users.get is deprecated, please use the new OctoPrint.access.users.get function instead [packed_client.js:18:307](http://localhost:5000/static/webassets/packed_client.js?673073c5)
Connected to the server [packed_core.js:1684:456](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Login mechanism: password [packed_core.js:267:35](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Triggering connect on all view models [packed_core.js:1689:240](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
Server (re)connect processed [packed_core.js:1694:5](http://localhost:5000/static/webassets/packed_core.js?f9f3ce36)
​```

[octoprint-systeminfo-20240910052616.zip](https://github.com/user-attachments/files/16939918/octoprint-systeminfo-20240910052616.zip)
[gettextattr.diff.txt](https://github.com/user-attachments/files/16939921/gettextattr.diff.txt)

foosel, 2024-09-10:
I have to admit that I have absolutely no clue why your custom binding doesn't work. From the looks of it, it seems that once that's been handled, knockout just stops applying bindings to inner elements, which is something I've so far not seen before.

However, what works is a simple `attr: { title: gettext($data.description) }`, no custom binding needed (and `gettext` is global within OctoPrint's runtime in any case, the fallback in the `gettext` binding is just in case someone copy-pastes that outside of OctoPrint).

But - looking at the specific case, I have no clue why I even went with `description` here for the row title and thus I've just switched it to the (untranslated) name of the permission. That's now pushed and ready for 1.11.0.

WildRikku, 2024-09-10:
> However, what works is a simple `attr: { title: gettext($data.description) }`, no custom binding needed (and `gettext` is global within OctoPrint's runtime in any case, the fallback in the `gettext` binding is just in case someone copy-pastes that outside of OctoPrint).

Just for my understanding, what exactly is `gettext` in this case? A JavaScript function? Jinja? I had a hard time figuring all the things (Jinja, Knockout, their i18n modules...) so I gave up on doing something in the template and instead created a binding because I understood how yours worked.

> But - looking at the specific case, I have no clue why I even went with `description` here for the row title and thus I've just switched it to the (untranslated) name of the permission. That's now pushed and ready for 1.11.0.

Well, that's simple and solves the issue. Thanks!

(Am I supposed to close it because I'm the OP or will your automation do that?)

foosel, 2024-09-11:
> Just for my understanding, what exactly is `gettext` in this case? A JavaScript function? Jinja?

A JavaScript function defined on the global `window` object that gets defined in the included `babel.js` file: https://github.com/OctoPrint/OctoPrint/blob/a8fff3930e3c3901bd560ca77656c281959134b3/src/octoprint/static/js/lib/babel.js#L65-L95

I admit that's one of these "inspect existing code to figure out it exists" things and could use more documentation. It's hard to balance all of these needs...

> (Am I supposed to close it because I'm the OP or will your automation do that?)

Automation *should* take care of it once the `maintenance` branch gets merged on `master` come the release of 1.11.0. Part of my release workflow is going through the issue tracker and manually closing everything that I tagged for the release and that for some reason didn't get auto-closed (usually thanks to me forgetting to include a `closes #<issuenumber>` in the commit).

So no further action needed here :)

WildRikku, 2024-09-11:
Thank you for the explanation and quick responses! :)

> I admit that's one of these "inspect existing code to figure out it exists" things and could use more documentation. It's hard to balance all of these needs...

Yeah, I thought I should probably write down what I figured out somewhere, maybe I will write up a little something for plugin devs or so.

foosel, 2024-09-11:
If you do, I'd be happy about a PR for the dev and/or plugin dev docs:

- https://github.com/OctoPrint/OctoPrint/tree/master/docs/development
- https://github.com/OctoPrint/OctoPrint/tree/master/docs/plugins

commit 439140f5 (2024-09-10): fix: use name of permission as title in list

Closes #5056
closed 2025-04-22 by foosel