OctoPrint/OctoPrint #5057: Queing Prints Gets Locked Up Under Certain Conditions

jneilliii, 2024-09-12:
### The problem

While printing if you go into plugin manager and install a plugin, it gets queued as expected. If you attempt to add another plugin after the initial queueing of the first plugin and install another it attempts to start install process and freezes. I think this is purely a UI issue, as a page refresh shows the additional plugins queued (based on button switching to dequeu).

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

Windows 10

### Printer model & used firmware incl. version

Virtual Printer

### Browser and version of browser, operating system running browser

Google Chrome Version 128.0.6613.137 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]

[octoprint-systeminfo-20240912145056.zip](https://github.com/user-attachments/files/16984369/octoprint-systeminfo-20240912145056.zip)

jneilliii, 2024-09-12:
appears to also cause the install queue to break, after cancelling the print not getting the queued count-down timer and attempting to manually install the queued prints using checkboxes and install selected button at the top hangs after the first plugin in the list gets installed. 

[image]

[image]

I'll try to take a look this weekend.

jneilliii, 2024-09-14:
Hmm..this may have been a red herring. I have been unable to reproduce the issue.

jneilliii, 2024-09-15:
issue appears to be due to checking to see if plugin is in queue.

what's being sent to remove...
```
{'command': 'install', 'url': 'https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip', 'dependency_links': False}
```

what's been added to self._queued_installs
```
{'url': 'https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip', 'dependency_links': False, 'from_repo': True, 'command': 'install'}
```

as you can see, the dictionary being dequeued is missing property `from_repo`. I suspect that was added for achievements, but haven't dug in that deep yet. Fix should be fairly easy and will be inbound shortly.

foosel, 2024-09-16:
Thanks for the analysis! Just to copy my response to your PR here:

> [...] I considered this bug to be a symptom of an underlying API issue. Having to know precisely with which parameters the install was enqueued in order to be able to remove it again caused this here. So I went with a subset check instead - we don't have to know all parameters, only some, and any items that contain these as a subset will be dequeued. Feels cleaner.

That not only fixes the current issue, but should also prevent anything similar from poppoing up in the future.

Ready for 1.11.0, but I'll also mark it as backport worthy should we require another 1.10.x bugfix release.

commit 1adaa63a (2024-09-16): fix[pgmr]: fix dequeue for install queue

Now it uses a subset of install properties to match
the items to remove. That should be a more solid
solution than having to remember to adjust things
whenever we need to modify the install command.

Closes #5057

See also #5061
commit f5a2dd47 (2024-09-16): fix[pgmr]: fix dequeue for install queue

Now it uses a subset of install properties to match
the items to remove. That should be a more solid
solution than having to remember to adjust things
whenever we need to modify the install command.

Closes #5057

See also #5061
closed 2024-11-05 by foosel