OctoPrint/OctoPrint #5059: Serial port configuration: "Hello" is send twice

NilsRo, 2024-09-13:
### The problem

1. See in the Terminal that "Hello" Command is send twice.
2. For better visibility change the  Hello-Command to something else. Did that for Klipper to restart firmware.

[image]

Can be reproduced with virtual printer. It is also interessting that after _Changing monitoring state from "Connecting" to "Operational"_ the original Hello-Command is send again and not FIRMWARE_RESTART. Or better I would no expect that it is send again.

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Yes

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

OctoPI 1.0.0

### Printer model & used firmware incl. version

Anycubic Mega S - Klipper v0.12.0-290-g14a83103c FIRMWARE_NAME:Klipper

### Browser and version of browser, operating system running browser

Windows 11 - Edge 128.0.2739.67

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[logs.txt](https://github.com/user-attachments/files/16990148/logs.txt)
[octoprint-systeminfo-20240913082832.zip](https://github.com/user-attachments/files/16990152/octoprint-systeminfo-20240913082832.zip)

Terminal log in Safe Mode with virtual printer:
Changing monitoring state from "Offline" to "Opening serial connection"
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: VIRTUAL(read_timeout=10.0,write_timeout=10.0,options={'enabled': True, '_config_version': 1}), starting monitor
**Send: FIRMWARE_RESTART
Recv: start
Send: FIRMWARE_RESTART
Recv: Marlin: Virtual Marlin!**
Send: FIRMWARE_RESTART
Recv: \x80
Recv: SD card ok
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok

jneilliii, 2024-09-13:
Scroll up, is wait for start on connect enabled or not? 

[image]

NilsRo, 2024-09-13:
It is disabled.

jneilliii, 2024-09-13:
Do you experience same issue if you enable it?

NilsRo, 2024-09-14:
Connection does not finish and stops with that output:

Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /tmp/printer, baudrate 250000
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x979193e8, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor

foosel, 2024-09-16:
The hello command being sent twice is actually intentional, and there's even a comment in the source that explains why:

https://github.com/OctoPrint/OctoPrint/blob/a8fff3930e3c3901bd560ca77656c281959134b3/src/octoprint/util/comm.py#L2298-L2304

So, in a nutshell, for some printers/setups, this second one is *crucial* to get the communication going, because the initial line received from the printer might still be garbage and eat up the obligatory acknowledgement response, so we need to trigger another response to make sure we are actually talking with the printer. That's also why the default hello command is `M110`. That the only command that we can send before we know anything about the printer on the other side and without knowing which line number it currently expects and if it even is enforcing line numbers.

After the initial handshake with those two hello comands, another `M110` is then sent which is the *explicit* line number reset and firmware info query:

https://github.com/OctoPrint/OctoPrint/blob/a8fff3930e3c3901bd560ca77656c281959134b3/src/octoprint/util/comm.py#L3726-L3736

So, in a nutshell, this isn't a bug. It's a workaround for some crappy setups out there that require some extra steps to get communication working, and for that reason this will have to stay in there.

What I could do here is allow to suppress the second iteration via some config setting, but that's all - the default behaviour has to stay the way things are, or there will be severe connection issues in the field with various setups.

foosel, 2024-09-16:
Ready for 1.11.0.

commit 2ce52c09 (2024-09-16): feat[comm]: allow to suppress 2nd hello command

Closes #5059
closed 2024-09-28 by NilsRo