OctoPrint/OctoPrint #5064: 'Software Update' missing from settings menu

Didel, 2024-09-20:
### The problem

The 'Software update' section is missing from the Settings menu. I would expect it to show up right between the 'Plugin Manager' and 'Announcements' links:

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

Version 1.10.2

### Operating system running OctoPrint

OctoPi 0.18.0

### Browser and version of browser, operating system running browser

Safari 18.0 (MacOS Sequoia 15.0)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

I feel like the system info bundle would expose information I don't necessarily think needs to be exposed 'out in the open'. If more information is needed, please let me know in which file the information can be found, then I'll see what I can do.

Probably relevant web-browser console output:

```
Error calling"onUserLoggedIn""on view model""ErrorTrackingViewModel"":"
"undefined is not an object (evaluating 'self.settings.settings.plugins.softwareupdate.octoprint_release_channel')
```

cp2004, 2024-09-20:
You have the "Software update" bundled plugin disabled. If you go to the plugin manager, you should be able to re-enable this from there.

Didel, 2024-09-20:
Interesting... 🤔 

I did not disable the software-update plugin myself, but I can indeed confirm that it is not enabled.

However, if I enable the plugin and reboot Octoprint, the plugin is disabled again. So it looks like something is preventing the Software Update plugin to run properly, which then disables the plugin again, if that does make sense?

cp2004, 2024-09-20:
Could you re-share a systeminfo bundle from after you have restarted OctoPrint now?

Didel, 2024-09-21:
> Could you re-share a systeminfo bundle from after you have restarted OctoPrint now?

Sure, this was after the 'Restart Octoprint' option from the menu was chosen and the system was back up online: 
[octoprint-systeminfo-20240921100836.zip](https://github.com/user-attachments/files/17083384/octoprint-systeminfo-20240921100836.zip)

foosel, 2024-09-21:
```
2024-09-20 15:41:58,371 - octoprint.plugin.core - ERROR - Error while parsing AST from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt/__init__.py
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 78, in parse_plugin_metadata
    root = ast.parse(f.read(), filename=path)
  File "/usr/lib/python3.7/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
ValueError: source code string cannot contain null bytes
2024-09-20 15:41:58,502 - octoprint.plugin.core - ERROR - Error while parsing AST from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 78, in parse_plugin_metadata
    root = ast.parse(f.read(), filename=path)
  File "/usr/lib/python3.7/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
ValueError: source code string cannot contain null bytes
``` 

You have a corrupted install. This might fix it: https://community.octoprint.org/t/how-to-reinstall-octoprint/36628

Didel, 2024-09-21:
Thanks @foosel for the reply and the great work on Octoprint!

Sadly, the reinstall also doesn't work:

```
octopi:~ $ ~/oprint/bin/pip install --force-reinstall OctoPrint==1.10.2 --no-cache-dir
Traceback (most recent call last):
  File "/home/pi/oprint/bin/pip", line 5, in <module>
    from pip._internal.cli.main import main
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/cli/main_parser.py", line 9, in <module>
    from pip._internal.build_env import get_runnable_pip
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/build_env.py", line 15, in <module>
    from pip._vendor.packaging.requirements import Requirement
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/packaging/requirements.py", line 23, in <module>
    from .markers import MARKER_EXPR, Marker
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/packaging/markers.py", line 23, in <module>
    from .specifiers import InvalidSpecifier, Specifier
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/packaging/specifiers.py", line 24, in <module>
    from .utils import canonicalize_version
  File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/packaging/utils.py", line 9, in <module>
    from .version import InvalidVersion, Version
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 857, in get_code
  File "<frozen importlib._bootstrap_external>", line 525, in _compile_bytecode
ValueError: bad marshal data (invalid reference)
pi@octopi:~ $ 

```

foosel, 2024-09-23:
That looks like rather far spread corruption then. I suggest to create a backup and flash a fresh card with a recent image, then restore the backup. That will hopefully make things work again.

Alternatively you might have luck recreating the virtual environment, but frankly that's a rather cumbersome progress that would also require you to reinstall all your plugins, and this looks like a somewhat broken storage and possibly a dying SD card, so I'd rather suggest to start with a fresh card.

In any case, that's not a bug in OctoPrint, that's a severely corrupted run time environment completely outside of its control. I'm thus closing this.

closed 2024-09-23 by foosel