OctoPrint/OctoPrint #5066: self._settings.get in plugins does not return custom settings created by on_settings_load

NilsRo, 2024-09-30:
### The problem

Using override of on_settings_load to add custom settings is reflected in e.g. Jinja2 Settings page but not taken care of in the _settings setter/getter.

1. Load a custom setting via on_settings_load
2. Try to use self._settings.get to retrieve the setting --> return null

https://github.com/jneilliii/OctoPrint-BackupScheduler/pull/26/commits/c066ef4d4cb90f58ab8dbc9a9e12ac0535c67707

### Did the issue persist even in safe mode?

No, it did not persist (it's an issue with one of your installed plugins, don't report here)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

OctoPrint Plugin interface issue

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

Windows (Development), OctoPI

### Printer model & used firmware incl. version

Anycubic Mega S - Klipper v0.12.0-290-g14a83103c FIRMWARE_NAME:Klipper

### Browser and version of browser, operating system running browser

Windows 11 - Edge 128.0.2739.67

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240930082928.zip](https://github.com/user-attachments/files/17185074/octoprint-systeminfo-20240930082928.zip)

It would be interesting if this is the intended behavior or a bug. If this is intended it should be clarified in the interface documentation somehow. https://docs.octoprint.org/en/master/plugins/mixins.html#octoprint.plugin.SettingsPlugin.on_settings_load

Perhaps it is this note pointing out to use preprocessors???:
`If you need more granular control here, e.g. over the used data types, you’ll need to override this method and iterate yourself over all your settings, using the proper retriever methods on the settings manager to retrieve the data in the correct format.`

https://github.com/jneilliii/OctoPrint-BackupScheduler/pull/26#discussion_r1780493229

foosel, 2024-10-01:
So, the way this works it that `self._settings.get` exclusively runs on what's in the defaults, in overlays or in `config.yaml` - in short, the actual configuration. `on_settings_(load|save)` is meant for making sure that a `bool` or `int` you have coming in *via the REST API* actually also gets saved like that (though preprocessors can be used for that as well), and it *can* also be used to inject custom additional stuff into what's available on the REST API that's not *actually* part of the config, but when it comes to the internal `self._settings` API, anything not defined someway in the defaults is not supported to be fetched through `get` (and IIRC also not even to be `set`). 

If you want custom data saved into `config.yaml`, you need to actually reserve some space for it (so to speak) in the defaults, otherwise `self._settings` won't know it.

The docs also speak of that difference, note how `on_settings_load` says

> Loads the settings for the plugin, called by the Settings API view in order to retrieve all settings from all plugins. Override this if you want to inject additional settings properties **that are not stored within OctoPrint’s configuration.**

(emphasis mine)

tldr; `self._settings` is your way to access OctoPrint's configuration. `on_settings_load` and `on_settings_save` allow you to change what get's sent/received via the REST API. The two are different, and `self._settings` doesn't care about what different behaviour you define for the REST API. The reason that manipulation option was implemented in the first place was to allow to interpret some internal settings for the frontend already and basically implement simple settings that can be manipulated more easily on the REST API before being translated into something more complicated internally (e.g. the release branch for OctoPrint in the Software Update Plugin).

I'll try to make this clearer in the documentation.

foosel, 2024-10-02:
The expectation is that usually what qualifies as a setting, a configuration, would actually be stored therein and not someplace else. Whereas anything that is not a setting but usage data or something like that would be stored elsewhere (e.g. in the data folder, managed fully by the plugin). The settings REST API is supposed to be used for the settings, so basically what is located in `config.yaml`. If you need further control than what that offers, and access to custom data stored elsewhere that doesn't qualify as settings (if it does, maybe it should be stored in the settings), you should implement your own API endpoint for fetching and manipulating that data, and not utilize the settings endpoint.

closed 2024-10-02 by NilsRo