OctoPrint/OctoPrint #5067: Maybe relax a little bit before sending M112 in some cases?

pointful-dave, 2024-10-02:
### Is your feature request related to a problem? Please describe.

The context is that I have been trying to find the source of print failures that looked like being due to various serial comms errors. I was finally present during a stop, and noticed that there was a loud, fast, almost full bed diagonal travel move just before the stop. I re-ran the print (this print has failed every time the last 6 times I've tried printing it) with travel speed set way lower and it completed. So it seems that despite multiple usb cables, power filters, and other things, it would seem that the problems were due to EMI.

The last stop is interesting:
```
024-10-02 12:57:58,081 - Send: N11365 G1 X37.27 Y182.898 E.01989*88
2024-10-02 12:57:58,090 - Recv: ok N11365 P1
2024-10-02 12:57:58,091 - Recv: ok N11362 P1 B15
2024-10-02 12:57:58,092 - Send: N11366 G1 F6939.371*83
2024-10-02 12:57:58,095 - Recv: ok N11363 P1 B15
2024-10-02 12:57:58,096 - Recv: ok N11364 P1 B15
2024-10-02 12:57:58,096 - Recv: ok N11365 P1 B15
2024-10-02 12:57:58,096 - Send: N11367 G1 X36.952 Y183.181 E.01817*102
2024-10-02 12:57:58,099 - Recv: ok N11366 P2 B15
2024-10-02 12:57:58,100 - Send: N11368 G1 F7200*70
2024-10-02 12:57:58,104 - Recv: Error:checksum mismat4 P1 B15
2024-10-02 12:57:58,105 - Recv: ok N11365 P1 B15
2024-10-02 12:57:58,105 - Recv: ok N11366 P2 B15
2024-10-02 12:57:58,105 - Recv: Error:checksum mismatch, Last Line: 11366
2024-10-02 12:57:58,106 - Send: N11369 G1 X36.262 Y183.845 E.03702*104
2024-10-02 12:57:58,107 - Recv: Resend: 11367
2024-10-02 12:57:58,107 - Recv: ok
2024-10-02 12:57:58,108 - Send: N11367 G1 X36.952 Y183.181 E.01817*102
2024-10-02 12:57:58,111 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 11366
2024-10-02 12:57:58,114 - Recv: Resend: 1136
2024-10-02 12:57:58,115 - Should resend line 1136 but no sufficient history is available, can't resend
2024-10-02 12:57:58,115 - Changing monitoring state from "Printing" to "Error"
2024-10-02 12:57:58,116 - Send: M112
2024-10-02 12:57:58,118 - Send: N11370 M112*37
2024-10-02 12:57:58,138 - Send: N11371 M104 T0 S0*36
2024-10-02 12:57:58,144 - Send: N11372 M140 S0*99
2024-10-02 12:57:58,150 - Changing monitoring state from "Error" to "Offline after error"
2024-10-02 12:57:58,160 - Connection closed, closing down monitor
```
this is about 10 lines after the travel move, and I notice the printer complaining about checksums, octoprint seeming to receive garbled messages "mismat4", and finally after the printer got something out of order after 11366 the resend line received is `Resend: 1136` which looks again like a serial error printer->octoprint.

So I don't believe there is any checksum for that direction (I guess there is not much that can be done if an error is detected like that), but looking at say https://github.com/OctoPrint/OctoPrint/blob/a8fff3930e3c3901bd560ca77656c281959134b3/src/octoprint/util/comm.py#L4483 this code, and the subsequent `abort the print & disconnect, there's nothing we can do to rescue it` I wonder: is that actually true? What if we were just to kind of assume there might have been something wrong with the command and resend the last line or something - the printer I believe will just keep requesting the correct line and if the issue is due to serial error then the print might still complete.

I wonder if there are some cases where a M112 is sent where it might be okay to just do nothing, or just resent the last line (at least a few times) in the hopes that it was a comms error? (Certainly there are other times where we shouldn't do that, but there might be several where the code it being a bit too pessimistic and failing prints that could continue)

Anyone want to weigh in on whether marlin will deal okay with that (it seems to me like it will: it will just keep asking to resend the last line if the checksum is wrong or the line number is wrong). Also maybe a list of places where this could be done and what might be an okay thing to do... maybe just resend the last line for which we don't have a recv response?

I am going to try to source a better shielded cable to see if it can survive the rough travel move, and I might also try out just sending the last line again in this particular case ("no history", though that has only been the cause twice)

### Describe the solution you'd like

Do not just send M112 on any error, some might be recoverable.

pointful-dave, 2024-10-02:
really? Okay, maybe this should not be a requirement for a "Request a feature" issue? Or maybe activate discussions?

[octoprint-systeminfo-20241002160756.zip](https://github.com/user-attachments/files/17220396/octoprint-systeminfo-20241002160756.zip)

jneilliii, 2024-10-02:
[image]

foosel, 2024-10-02:
> Okay, maybe this should not be a requirement for a "Request a feature" issue?

It actually isn't, but you removed the `[Request]` line from the title, and alas, that meant this wasn't detected as a feature request. I'll take a look at the reason for that, because the template *should* also set the label, but maybe that's a case of a missing feature in GitHub's issue tracker. IN any case, I've fixed this manually now.

---

Ok, so, the problem with messing around with the behaviour here is, it's removing some safety margin. Currently, when (for whatever reason) the communication is so bad that even resends no longer work correctly, the approach is to *stop anything that produces heat and movement* and disconnect. The idea behind that is simple, reduce the risk of anything starting to burn or break in other ways, then switch to a state where the user is absolutely clear that something went wrong that they need to investigate and if possible fix.

Anything messing with this and adding additional timeouts or such increases the likeliness that something DOES in fact go terribly wrong.

If you printer communication is suffering from EMI so much that it requires resends and those don't even come through correctly, that problem needs to be solved. That is not something that OctoPrint should try to hide and/or work around by trying to be intelligent about a situation it has absolutely no way to understand fully, and if it did this then whoever opened the bug report about this would be completely and 100% right.

I understand that EMI issues like that are frustrating, but that's what needs to get fixed here, that's what should be the focus.

closed 2024-10-02 by foosel

evilDave, 2024-10-02:
Ah right my mistake editing the topic line.

I know about the "disable disconnect" and did not want to do that for the same reasons as pointed out by @foosel above.

My thinking was that perhaps serial errors were just a fact of life for some environments and it might be nice to be able to have a setting to not disconnect in cases where it "looks like" it was a serial error. Surely the printer requesting a weird line number is no cause for immediate alarm? But i suppose the ones where it is disconnecting due to "any error" is not going to be detectable and could well be something that should result in disconnect and M112. Maybe if there could be a checksum on the incoming message... but then again maybe the reason for that is that the usb cable is on fire :)

foosel, 2024-10-02:
> Surely the printer requesting a weird line number is no cause for immediate alarm?

I would agree with you if I hadn't seen what vendors do to firmware over the past decade, and sadly, ANYTHING that looks wrong in the communication is cause for alarm :/ The firmware variants out there are sometimes such a mess, it's really a good idea to just be very very cautious.

evilDave, 2024-10-02:
Yeh I was wondering would it be a mistake to leave the safety concerns up to the firmware, but it probably would. (I am using mriscoc firmware which I would trust quite a lot to have the control of the printer "under control" but that would be forgetting the range that octoprint is dealing with)

(I see now that this would have been better off as a post in the forum, apologies)

foosel, 2024-10-02:
All good, it's understandable that you opened that as a request here, reality just gets into the way here once more 😅

jneilliii, 2024-10-02:
> I am using mriscoc firmware

Interestingly enough, the screenshot I shared is directly from his wiki on how to configure OctoPrint. I agree with foosel that this is a bad approach though. 

https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#error-handling