OctoPrint/OctoPrint #5068: .code files from SD card non listed in Files section

mnolletti69, 2024-10-16:
### The problem

I have a strange issue with the reading of files from the SD card:
- Long filenames with extension .gco are listed
- Long filenames with extension .gcode are NOT listed
- If I only change the extension from .gcode to .gco it will be listed
- In terminal "M20 L T" list ALL files with short and long names
Executing a command **dir /X** in a command prompt I noticed that the short 8.3 names now contains ~ character also in the extension (MyLongFileName.gcode -> MYLONG~1.GC~)
This is my guess:
When long names are enabled, the parsing of the long names is executed correctly from the "M20 L T" command but the filtering for printables files is executed on the 8.3 name. If this is the case then .CG~ files are filtered out as non valid printable files.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

1.0.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

### Printer model & used firmware incl. version

Ender 3 with Marlin 2.1.2.4 (also with 1.1.9)

### Browser and version of browser, operating system running browser

All browser, Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241017000225.zip](https://github.com/user-attachments/files/17403191/octoprint-systeminfo-20241017000225.zip)

mnolletti69, 2024-10-16:
Errata corrige: (MyLongFileName.gcode -> MYLONG~1.GC~)

mnolletti69, 2024-10-16:
Errata corrige 2: The renderer cuts the double "~" character so I use [tilde] instead:
 (MyLongFileName.gcode -> MYLONG[tilde]1.GC[tilde])

foosel, 2024-10-17:
> the parsing of the long names is executed correctly from the "M20 L T" command but the filtering for printables files is executed on the 8.3 name

Yes, and the 8.3 names is also the only ones that the firmware will understand. The long names are basically just wallpapered over the 8.3 ones for aesthetic reasons. I honestly so far hadn't seen `.gcode` get turned into `.gc~`, but extending the filter to treat that as a valid file is simple enough, so I'll pencil that in for 1.11.0.

foosel, 2024-10-17:
Ready for 1.11.0.

timur-tabi, 2024-10-31:
I have a similar problem.  With my Prusa MK3.5s, Octoprint does not recognize STAR_0~1.GCO as a gcode file.  

Details in this thread: https://community.octoprint.org/t/only-show-files-stored-on-sd-doesnt-show-any-files/61051

commit 69326e35 (2024-10-17): fix: recognize .gc~ as gcode extension

In 8.3 format, `foo.gcode` will turn into `foo.gc~`. Support
that.

Closes #5068
closed 2025-04-22 by foosel