OctoPrint/OctoPrint #5071: Incorrect SHA256 of 1.10.2 tarball

Beep6581, 2024-10-20:
### The problem

Trying to install octoprint-1.10.2 in Manjaro. Seems the .tar.gz file has the wrong SHA256.

```
yay -S octoprint
AUR Explicit (1): octoprint-1.10.2-1
:: (1/1) Downloaded PKGBUILD: octoprint
  1 octoprint                                (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> A
:: Deleting (1/1): /home/beep/.cache/yay/octoprint
HEAD is now at 1c3aa5a 1.10.2
  1 octoprint                                (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
==> Making package: octoprint 1.10.2-1 (sön 20 okt 2024 23:12:31)
==> Retrieving sources...
  -> Downloading octoprint-1.10.2.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 5823k    0 5823k    0     0  1172k      0 --:--:--  0:00:04 --:--:-- 1394k
  -> Found octoprint.sysusers
  -> Found octoprint.tmpfiles
  -> Found octoprint.service
  -> Found octoprint-serve
  -> Found octoprint.conf
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    octoprint-1.10.2.tar.gz ... FAILED
    octoprint.sysusers ... Passed
    octoprint.tmpfiles ... Passed
    octoprint.service ... Passed
    octoprint-serve ... Passed
    octoprint.conf ... Passed
==> ERROR: One or more files did not pass the validity check!
 -> error downloading sources: /home/beep/.cache/yay/octoprint 
         context: exit status 1 

cd .cache/yay/octoprint
grep -A8 source= PKGBUILD
source=("${pkgname}-${pkgver}.tar.gz::https://github.com/OctoPrint/OctoPrint/archive/refs/tags/${pkgver}.tar.gz"
                octoprint.sysusers
                octoprint.tmpfiles
                octoprint.service
                octoprint-serve
                octoprint.conf
                )
options=(!distcc !ccache)
sha256sums=('460e6abedb70b521f895126a53b3a38ae47b97185b6d5e0934dc9257be72a122'

ls -sh octoprint-1.10.2.tar.gz
5,7M octoprint-1.10.2.tar.gz
sha256sum octoprint-1.10.2.tar.gz
43de021565107d56dea30f5c0c0efcc779c0d9b2bd4665ea791e29a69f996080  octoprint-1.10.2.tar.gz

rm octoprint-1.10.2.tar.gz
wget "https://github.com/OctoPrint/OctoPrint/archive/refs/tags/1.10.2.tar.gz"
--2024-10-20 23:16:51--  https://github.com/OctoPrint/OctoPrint/archive/refs/tags/1.10.2.tar.gz
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving github.com (github.com)... 4.225.11.194
Connecting to github.com (github.com)|4.225.11.194|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/OctoPrint/OctoPrint/tar.gz/refs/tags/1.10.2 [following]
--2024-10-20 23:16:52--  https://codeload.github.com/OctoPrint/OctoPrint/tar.gz/refs/tags/1.10.2
Resolving codeload.github.com (codeload.github.com)... 4.225.11.198
Connecting to codeload.github.com (codeload.github.com)|4.225.11.198|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘1.10.2.tar.gz’

1.10.2.tar.gz                              [                          <=>                                                ]   5,69M   968KB/s    in 5,5s    

2024-10-20 23:16:58 (1,03 MB/s) - ‘1.10.2.tar.gz’ saved [5962894]

ls -sh 1.10.2.tar.gz
5,7M 1.10.2.tar.gz
sha256sum 1.10.2.tar.gz
43de021565107d56dea30f5c0c0efcc779c0d9b2bd4665ea791e29a69f996080  1.10.2.tar.gz

rm -v octoprint-1.10.2.tar.gz 1.10.2.tar.gz

wget https://github.com/OctoPrint/OctoPrint/releases/download/1.10.2/octoprint-1.10.2.tar.gz
(...)
ls -sh octoprint-1.10.2.tar.gz
3,0M octoprint-1.10.2.tar.gz
sha256sum octoprint-1.10.2.tar.gz
9661bbf811137f9ff7fdf3ede42313dc58f76401e96cd017115a9d47f02a2671  octoprint-1.10.2.tar.gz

rm -v octoprint-1.10.2.tar.gz 1.10.2.tar.gz
wget https://github.com/OctoPrint/OctoPrint/releases/download/1.10.2/OctoPrint-1.10.2.source.tar.gz
(...)
ls -sh OctoPrint-1.10.2.source.tar.gz
5,7M OctoPrint-1.10.2.source.tar.gz
sha256sum OctoPrint-1.10.2.source.tar.gz
2c56f172a9e28ad1e713c893c35a9cf541521405b85ba8ebdbbcac458c5621bf  OctoPrint-1.10.2.source.tar.gz
```

Note that
`https://github.com/OctoPrint/OctoPrint/archive/refs/tags/1.10.2.tar.gz`
downloads a different file to
`https://github.com/OctoPrint/OctoPrint/releases/download/1.10.2/octoprint-1.10.2.tar.gz`

Seems that changing hashes have occurred several times before:
https://aur.archlinux.org/packages/octoprint

Are you altering tarballs but using the same filename and version?

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

n/a

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

Manjaro

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2024-10-20:
On the releases page of this repository (for 1.10.2) there are 3 files generated by build actions:

* OctoPrint-1.10.2-py2.py3-none-any.whl
* OctoPrint-1.10.2.source.tar.gz
* octoprint-1.10.2.tar.gz

There's also sha512sums.txt file, which matches the files when I download them now.
There's no sha256 provided, only in the [repository you're trying to use](https://github.com/pa314159/octoprint-arch) - we can't control what 3rd parties decide they will do when distributing OctoPrint. It looks like this repo tries to use the GitHub-generated asset, which we can't control, and it may be subject to changes as/when the repository changes (or as GitHub feel like it). Hence why there are built release assets, which are intended to be stable from 1.10.0. Background in https://github.com/OctoPrint/OctoPrint/issues/4908 - suggest they use these instead.

> Note that
https://github.com/OctoPrint/OctoPrint/archive/refs/tags/1.10.2.tar.gz
downloads a different file to
https://github.com/OctoPrint/OctoPrint/releases/download/1.10.2/octoprint-1.10.2.tar.gz

This is expected, because the first is automatically generated tarball of the repository by GitHub (what you get if you click "source code" on the release), the second is created via build actions using setuptools, which you could use to install OctoPrint.

closed 2024-10-20 by cp2004

Beep6581, 2024-10-22:
Thank you, reported to AUR package maintainer:
https://github.com/pa314159/octoprint-arch/issues/8