OctoPrint/OctoPrint #5072: [Request] Filter option for files list

pLazyN, 2024-10-22:
### Is your feature request related to a problem? Please describe.

Hello,

I hope I did not miss it but since multiple users upload files to our octoprint server I would like to be able to filter them e.g. by username to review all my files, delete old unnecessary one and put other into folders.
Especially when sending the files from within PrusaSilcer the files list becomes quite messy.

Kind regards,
Martin

### Describe the solution you'd like

Maybe search operators like "user:" within the search bar? 

### Describe alternatives you've considered

Loading each files manually to see who uploaded the file.

I see there is a "apply filters to search results" which does not seem to to anything. 
At least I could not find where to set those filters

foosel, 2024-11-19:
Totall forgot to comment here, but with 1.11.0 this will be possible with an additional search parameter `user:<name>`, e.g. `user:foosel test` will find all files uploaded by user foosel that contain `test` in their name.

Sadly this information about who uploaded a file was not stored so far however, so this will only work with new uploads.

pLazyN, 2024-11-20:
Thanks a lot Gina! 

One question: If I load an existing file and look at the "state" tile there the information who uploaded the file is listed. So doesn't that mean that it has to be stored in the file already?

jneilliii, 2024-11-20:
I think that is the user that started the print job, not uploaded the file.

foosel, 2024-11-20:
What @jneilliii said is correct, that's the one who started the print.

foosel, 2025-04-22:
Closing this as 1.11.0 has just been released

closed 2025-04-22 by foosel